schreibfaul1/ESP32-audioI2S #802: ESP32-S3 and PCM5102 not working

bncdemo, 2024-08-08:
I am running the example on ESP32-S3-Zero with audio.connecttohost(...)
The code run without visible exception or error, but simply there is no sound.
What could be the problem? 🤔

FatherMarco1971, 2024-08-09:
You give very ppoor informations anyway PCM5102 can be tricky..
I suggest that you try with FLT:Low, DEMP: Low, XSMT: High, FMT: Low

bncdemo, 2024-08-10:
> You give very ppoor informations anyway PCM5102 can be tricky.. I suggest that you try with FLT:Low, DEMP: Low, XSMT: High, FMT: Low

Hi, what other info should I write? 
The board info can be seen here: waveshare.com/wiki/ESP32-S3-Zero
I am wondering if the memory is too small? - but why there is no exception anywhere? 😳
Is there an example which shows (print to the terminal) the progress of the played audio?

schreibfaul1, 2024-08-10:
The board is absolutely suitable. Built-in 512KB of SRAM and 384KB ROM, onboard 4MB Flash memory and 2MB OCTAL PSRAM.
In your development environment, the settings should be something like this:
PSRAM: OPI
Partition: Huge App
CPU Frequency 240MHz

For local files (from the SD card) there is 
`getAudioFileDuration()'` and `getAudioCurrentTime()` which you would then have to call periodically

bncdemo, 2024-08-10:
I am using audio.connecttohost.

Is there a buffering settings somewhere, what can cause a 10 seconds play, 2-3 secs cut (pre-buffering?) repeated cycle?

FatherMarco1971, 2024-08-10:
Aare you Shure connection are right?
Have you tried o follow my suggestion?

bncdemo, 2024-08-11:
> Aare you Shure connection are right? Have you tried o follow my suggestion?

Yes, same, except DEMP not set.
Is there no way to query any audio related info when using connecttohost? Stream speed/name, elapsed playing time and similar infos? 🤔

closed 2024-09-25 by github-actions[bot]