schreibfaul1/ESP32-audioI2S #803: Starting audio from a position does not work in connecttoFS

kungfupizza, 2024-08-12:
Here is what I am trying as flow,
Start playing a song with, `audio.connecttoFS(SD, "xyz.mp3");`

Stop after some time and get the seek, `uint32_t seek = audio.getFilePos();`

Restart the song from a seek, `audio.connecttoFS(SD, "xyz.mp3", seek);` here seek value is `14756`

The song starts from start instead of starting from that seek value. 
Some useful prints from `Audio::processLocalFile`
`
processLocalFile(): m_audioDataStart 138
processLocalFile(): m_resumeFilePos 14756
processLocalFile(): Corrected m_resumeFilePos 22706
`

What am I doing wrong?

schreibfaul1, 2024-08-12:
The file pointer is not synchronized with the audio currently being played. audio.stopSong() returns a calculated value that you can use in connecttoFS().

kungfupizza, 2024-08-14:
Thank you! It has gotten better with `stopSong` BUT still does not start from the point it left rather starts from just a few seconds from the start.

closed 2024-09-28 by github-actions[bot]