schreibfaul1/ESP32-audioI2S #806: Play without blocking further code

philxws692, 2024-08-15:
Hey there,

first of all great project. I got a little question as I am not finding anything in the examples which could help me.
I am trying to play a sound while blinking an led.
Is there any possibility? Since the `audio.loop()` is playing the audio while being placed inside a loop, I did not come up with a solution. 
So the example of blinking an LED is very basic, but I want to build an accessoires for a board game which needs sound but also the execution of further code should not be blocked while the audio is playing.

Maybe I have a "knot in my brain", so any help would be very much appreciated 😀

Best regards
Philipp

schreibfaul1, 2024-08-15:
Hello Philipp, the audio.loop() must not be interrupted for too long. You could solve your problem like this:
````c++
uint32_t t = 0;
bool led = false;
...
void setup(){
    ...
    t = millis() + 1000;
}

void loop(){
    ...
    if(t < millis()){
        t = millis() + 1000;
        led = !led;
        if(led) log_i("led on");
        else log_i("led off");
    }
    ...
    audio.loop();
}
````

philxws692, 2024-08-16:
Well didn't think it's that easy, thank you very much 😄

closed 2024-08-16 by philxws692