schreibfaul1/ESP32-audioI2S #809: About synchronizing lyrics

LemonStudio, 2024-08-19:
Hello, thanks for you library. 

I want to make a music player. It has an OLED display screen and a PCM5102 audio chip. I want to display the title and lyrics on the screen. At present, I have solved the problem of displaying titles by reading Wiki, but I don't know how to display lyrics. How can I solve this problem? (There are synchronized lyrics in my audio file) 

Thank you very much.
Lemon Studio.

schreibfaul1, 2024-08-19:
Here is an example of this: https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Synchronised%20lyrics , mp3 files with embedded lyrics are rarely found. The audioI2S library does not read in the lyrics data, but these are loaded from the SD card. In the example, the lyrics are read out according to the time stamp and displayed in the serial monitor.

LemonStudio, 2024-08-20:
OK, I Know, Thank you very much.

closed 2024-08-20 by LemonStudio