schreibfaul1/ESP32-audioI2S #810: Ho to clear the Audio buffer dymanically

monupurohit, 2024-08-19:
I need to handle cashing, how to clear cashing/ need to clear buffer.

My Internet radio is not playing online stream.
When pressing EN button for multiple time it plays.
Unable to identify the issue

#include <WiFi.h>
#include <WiFiManager.h>
#include <HTTPClient.h>
#include <Arduino_JSON.h>
#include <Audio.h>
#include <EEPROM.h>
#include <WebServer.h>

#define I2S_DOUT  25
#define I2S_BCLK  27
#define I2S_LRC   26
#define MAX_LOG_SIZE 1024    // Maximum size of the log buffer

WebServer server(80); // Create WebServer object on port 80

Audio audio;
int subscription = 0; // Default value for subscription status
String logBuffer = ""; // Buffer to store logs

String serverRead = "https://*****.amazonaws.com/Test//SubscriptionStatus?Period=MAC=";

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    
    WiFiManager wm;
    if(!wm.autoConnect("Bhakti2","1q2w3e4r")) {
        Serial.println("Failed to connect");
        addToLogBuffer("Failed to connect to WiFi");
        return;
    }
    Serial.println("Connected to WiFi");
    addToLogBuffer("Connected to WiFi");
    delay(3000);

    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(100);
    // Delay to allow audio setup, no direct buffer size method available
    delay(3500); 

    subscription = receiveData(); // Check subscription status

    // Set up the web server route
    server.on("/", HTTP_GET, [](){
        server.send(200, "text/plain", logBuffer);
    });

    server.begin(); // Start the server

    // Connect to appropriate host based on subscription status
    if (subscription > 0) {
        Serial.println("User Subscription is Active");
        addToLogBuffer("User Subscription is Active");
        audio.connecttohost("http://vis.media-ice.musicradio.com/CapitalMP3");
    } else {
        Serial.println("User Subscription is NOT Active");
        addToLogBuffer("User Subscription is NOT Active");
        audio.connecttohost("http://vis.media-ice.musicradio.com/BigTop40");
    }
}

void loop() {
    server.handleClient(); // Handle incoming client requests
    audio.loop(); // Process audio

    // Flush and clear the Serial buffer to prevent overflow
    Serial.flush();
}

void addToLogBuffer(String message) {
    logBuffer += message + "\n";

    // Trim the log buffer if it exceeds the maximum size
    if (logBuffer.length() > MAX_LOG_SIZE) {
        int excessLength = logBuffer.length() - MAX_LOG_SIZE;
        logBuffer = logBuffer.substring(excessLength);
    }
}

void sendData(int dta) {
    HTTPClient http;
    String url = serverName + "&field1=" + dta;
    http.begin(url.c_str());
    int httpResponseCode = http.GET();
    if (httpResponseCode > 0) {
        Serial.print("HTTP Response code: ");
        Serial.println(httpResponseCode);
        addToLogBuffer("HTTP Response code: " + String(httpResponseCode));
    } else {
        Serial.print("Error code: ");
        Serial.println(httpResponseCode);
        addToLogBuffer("Error code: " + String(httpResponseCode));
    }
    http.end();
}

int receiveData() {
    HTTPClient http;
    String url = serverRead + WiFi.macAddress();
    http.begin(url.c_str());
    int httpResponseCode = http.GET();
    if (httpResponseCode > 0) {
        String payload = http.getString();
        JSONVar myObject = JSON.parse(payload);
        if (JSON.typeof(myObject) == "undefined") {
            Serial.println("Parsing input failed!");
            addToLogBuffer("Parsing input failed!");
            return 0;
        }
        int subscriptionStatus = (int)myObject["SubscriptionStatus"]; // Assuming the JSON field name
        addToLogBuffer("Subscription Status received: " + String(subscriptionStatus));
        return subscriptionStatus; // Should be 0 or 1 based on your API response
    } else {
        Serial.print("Receive Error code: ");
        Serial.println(httpResponseCode);
        addToLogBuffer("Receive Error code: " + String(httpResponseCode));
        return 0; // Default to 0 if there's an error
    }
    http.end();
}

monupurohit, 2024-08-19:
This code works pretty well, however somtime media is not getting played
when trying to power on/Off multiple time or Pressing EN switch multiple time, this plays.

how to debug if this is network issue or some cache issue causing not to play the media stream

monupurohit, 2024-09-19:
Any assistance?

closed 2024-11-04 by github-actions[bot]

monupurohit, 2025-01-12:
can anyone look on this ??

schreibfaul1, 2025-01-15:
There is no function that does this. The buffer is reset by setting the write and read pointer to the start of the buffer.