schreibfaul1/ESP32-audioI2S #811: text section exceeds available space in board?

hzkincony, 2024-08-19:
i use ESP32-S3 N16R8 , but when i compile, it showed:
Sketch uses 1340981 bytes (102%) of program storage space. Maximum is 1310720 bytes.
text section exceeds available space in board
Global variables use 46284 bytes (14%) of dynamic memory, leaving 281396 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board ESP32S3 Dev Module.

how to solve? thanks

schreibfaul1, 2024-08-19:
You have 16MB flash, which is still plenty of space. The easiest method is to change the partition table. You can use "Huge App".

skylai82, 2024-09-20:
[image]
How to reduce Audio.h space?

skylai82, 2024-09-20:
> ![image](https://private-user-images.githubusercontent.com/37041202/369313095-31694b64-9a57-41c4-b67d-a7fbbc76f836.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

[image]
Can I edit my partition scheme?

closed 2024-11-04 by github-actions[bot]