schreibfaul1/ESP32-audioI2S #817: Distorted sound on AAC

hevet, 2024-08-25:
From this version https://github.com/schreibfaul1/ESP32-audioI2S/commit/49892f7d11a83cde03f20aa2e36c9d87f7c22f42 e.g. on this station broadcasting in AAC https://n13a-eu.rcs.revma.com/7vhveccrnquvv the sound is distorted. It can be compared as if some voiced sounds had a lisp.

stef-ladefense, 2024-08-31:
hi,
have some distorted with https://jazzradio.ice.infomaniak.ch/jazzradio-high.aac

hevet, 2024-08-31:
@schreibfaul1 can something be done about it?

schreibfaul1, 2024-09-01:
I cannot observe this with the test sketch on the readme. Both URLs work with the ESP32 and ESP32-S3.
Is there anything conspicuous in the log?

hevet, 2024-09-01:
@schreibfaul1 Streams work only the sound quality deteriorates. I can't describe it exactly but it's more lisping.

philippedc, 2024-09-01:
Hi, i get similar issues, so i tested different releases to find the good one

hevet, 2024-09-01:
I'm still using the older version, the idea is to fix the problem in newer versions.

stef-ladefense, 2024-09-01:
I did the test with the sketch of the readme. there are micro cuts with the master version (zip of this morning) I have the impression of listening to an old vinyl :)

on the other hand, if I replace the library with version 3.012 of July 29, it works without any cut as I already said

stef-ladefense, 2024-09-01:
ps: mp3 decoding still works. I notice that aac decoding uses more resources than with version 3.012 (f)
I have functions of my radioweb application that slow down with the current master version and not with the old one

schreibfaul1, 2024-09-01:
Hello @stef-ladefense,
 the old Helix AAC decoder has been replaced by the [faad2](https://github.com/knik0/faad2) decoder. faad2 enables parametric stereo and is still being developed for Linux applications.  However, faad2 requires more resources and pushes the ESP32 to its limits. Nevertheless, good playback is possible.
There are two tasks that are responsible for this. This is the Arduino loop task that controls audio.loop() and thus places the data from WiFi in the buffer. And an internal task that takes the data from the buffer and decodes it. This internal task is on core 0 and can be changed with audio.setAudioTaskCore();. It may help to swap the cores.

hevet, 2024-09-01:
@schreibfaul1 On ESP32s3 it's the same, on miniwebradio there's the same problem. The problem appeared from the version I gave in the first post, and in the previous one there was already a new AAC decoder and it worked fine.

hevet, 2024-09-01:
Below, for comparison, are two recordings where you can hear the difference in decoding.
https://drive.google.com/drive/folders/1JFErZHVXK7qP3-V4lZxGLhvVtV1tBDCF

stef-ladefense, 2024-09-01:
thanks for your clarification schreibfaul1,

on my application, I played with the priorities and cores to see.
(in xTaskCreateStaticPinnedToCore).
and still on https://jazzradio.ice.infomaniak.ch/jazzradio-high.aac.

on core 1, I need priority 22 to have a correct sound but I no longer have any other tasks that work normally.
on core 0, switching to priority 19 works for me with a noticeable overall slowdown.

IMPORTANT NOTE: the playback issues do not concern all aac streams.

here is another French aac stream that does not work correctly: http://ouifm.ice.infomaniak.ch/ouifm-high.aac

but the following aac streams have no listening issues with the master version (in priority 2) :
http://europe1.lmn.fm/europe1.aac
http://europe2.lmn.fm/europe2.aac
http://icecast.radiofrance.fr/fip-midfi.aac
http://icecast.radiofrance.fr/fb1071-midfi.aac
http://icecast.radiofrance.fr/franceculture-midfi.aac
http://icecast.radiofrance.fr/franceinfo-midfi.aac
http://icecast.radiofrance.fr/franceinter-midfi.aac
http://icecast.radiofrance.fr/francemusique-midfi.aac
http://icecast.skyrock.net/s/natio_aac_64k
http://rfm.lmn.fm/rfm.aac

it's very strange that these 2 streams load the cpu completely.
and the other streams work perfectly...
i'm not good enough to understand why these 2 streams at my place increase the cpu load to its limits.

while waiting for better, i'll replace the 2 aac streams that don't work with their mp3 streams.

you're talking about high resources for the new faad2, but here my other aac streams work normally without loading the cpu beyond limits.

if i have time, i'll leave a message on the faad2 github with the two twisted aac streams.

schreibfaul1, 2024-09-01:
Well, the aac formats are not all identical. I see you are using an ESP32-S3. SBR and PS are always active for the S3. To save computing time, you can deactivate them for tests. Simply comment it out in neaacdec.h 
````
    #define SBR_DEC /* Allow decoding of SBR (spectral band replication) */
    #define PS_DEC /* Allow decoding of PS (parametric stereo) */
````

stef-ladefense, 2024-09-01:
> Well, the aac formats are not all identical. I see you are using an ESP32-S3. SBR and PS are always active for the S3. To save computing time, you can deactivate them for tests. Simply comment it [...]

when at least one of the two is disabled, it works perfectly.
I disabled both, because on a small 8cm mono speaker it doesn't bring anything.

where does the content of your aac_decoder folder come from?
because I would like to add two routines Set_xxx() to choose the SBR_DEC and PS_DEC configuration directly accessible in the program for the future.

thank you very much

hevet, 2024-09-01:
It doesn't change anything for me, some stations in AAC still have worse quality.

stef-ladefense, 2024-09-01:
> Cela ne change rien pour moi, certaines stations en AAC ont encore une qualité pire.

in the neaacdec.h file ?

VaAndCob, 2024-09-03:
I use ESP32 Wrover with PSram and ES8388 codec (it's AI Thinker Audio Kit V2.2)
all sound sources work well,  except playing .m4a file.  it played but did not sound and very short like play and suddenly stopped. this is what I see in the callback function cab read file info,
but no sound at all.

info        PSRAM found, inputBufferSize: 638965 bytes
info        buffers freed, free Heap: 183772 bytes
info        Reading file: "/response.m4a"
info        AACDecoder has been initialized, free Heap: 182860 bytes , free stack 3308 DWORDs
info        ch; 1, bps: 16, sr: 16000
info        unknown Audio Type 0
info        max bitrate: 1073743106
info        avr bitrate: 336070145
info        unknown ObjectType e, stop
info        Closing audio file "response.m4a"
info        Sampling Frequency: 44100
info        ch; 1, bps: 16, sr: 16000

hevet, 2024-09-09:
You're confusing the concepts. The problem I described is that some stations broadcast in AAC have audio distortion, not that some don't work. @schreibfaul1 do you plan to fix this issue in any of the next library versions?

closed 2024-09-15 by hevet

stef-ladefense, 2024-09-22:
hello, this modification is a good thing for a simple ESP32.
but some flows, as mentioned above, are too much for an ESP32S3.

this modification should rather be controlled via an option audio.SBR_DEC(true or false) and audio.PS_DEC()...

what do you think?

https://github.com/schreibfaul1/ESP32-audioI2S/commit/75485f841fb3f6de21bfe0bbefbe18d5f90acecc

schreibfaul1, 2024-09-24:
The ESP32 does not manage SBR and PS. The changes have been inserted so far.

stef-ladefense, 2024-09-25:
indeed, and creating a function instead of modifying the source would be a good thing.