schreibfaul1/ESP32-audioI2S #821: Is there a compatibility problem between audioI2S and c++ String manipulation?

chard62, 2024-08-28:
c++ noob here, sorry if this is the wrong place.
If this is the right place, can you please offer some suggestions?
If this is the wrong place, please suggest the best place to start.

I have a simple radio that is working, thanks to your great sample code.  This is on an ESP32-S3-Zero (Waveshare).  I'm using Platform IO as the IDE (in VS Code).

Now I am trying to save the station name and stream title in a string variable, inside the `audio_showstation() `and `audio_showstreamtitle()` functions.  Here's a small snippet:

> char *station_name;
> char *stream_title;
> 
> void audio_showstation(const char *info)
> {
>   Serial.print("Station -      ");
>   Serial.println(info);
>   // Copy the value of "info" (the station name) into a char* variable
>   // this causes the ESP32 to go into a bootloop:
>   //strcpy(station_name, info);
> 
>   // given that strncopy causes a bootloop, trying to do the copy using += in a loop
>   // the problem we see now, besides the c++ warning messages, is the Serial port displays partial 
>   // error messages from audioI2S  - we can't see the whole message, probably because of timing issues
> 
>   //station_name = "";  // c++ complains that "ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]"
>   //for(int i = 0; (i < strlen(info)) && (i < 40); i++){
>   //  station_name += info[i];
>   //  Serial.println(station_name);
>   //}
> 
> }

You can see in the comments that when I use strcpy(), the ESP32 goes into a bootloop.
When I try `station_name = "";,` c++ complains.
If I just go ahead and compile with the loop code (last 5 lines), the serial port displays only partial error messages coming from audioI2S.

platformio.ini:

> [env:esp32-s3-fh4r2]
> platform = espressif32
> board = esp32-s3-fh4r2
> framework = arduino
> upload_port = COM24
> monitor_port = COM10
> monitor_speed = 115200
> lib_deps = 
>     khoih-prog/ESPAsync_WiFiManager@^1.15.1
>     esphome/ESP32-audioI2S@^2.0.7

schreibfaul1, 2024-08-28:
The easiest way is as follows:

`````c++
// global
String stationName = "";
_f_newStationName = false

void loop(){
    ...
    if(_f_newStationName){ // station has changed
        _f_newStationName = false;
        if(stationName.startsWith,.....
        ......
    }
}

void audio_showstation(const char* info) {
    if(!info) return;
    stationName = info;
    SerialPrintfln("StationName: %s", info);
    _f_newStationName = true;
}

chard62, 2024-08-30:
YES!  Thank you so much.  The problem was that I was using char* when I should have used String!  Being a c++ noob, I thought they were equivalent.  Now I know something new, thanks to your help!

chard62, 2024-08-30:
Closing now.

closed 2024-08-30 by chard62