schreibfaul1/ESP32-audioI2S #822: Possibility to use 2 Audio at the same time

AntoineLep, 2024-08-28:
Hi,

I would like to use 2 audios at the same time to play a WAV voice feedback when I do a specific action while listening to a main audio MP3 file but It actually doesn't work when declaring:
```c++
Audio audio;
Audio audio2;
```

I got the following errors: 
```
E (1231) i2s_common: i2s_new_channel(812): no available channel found
E (1231) i2s_std: i2s_channel_init_std_mode(199): input parameter 'handle' is NULL
E (1233) i2s_common: i2s_channel_enable(999): input parameter 'handle' is NULL
```

Then one audio can't play at all.

I've tried using a single Audio but it can't handle audio file switch quick enough for me. MP3 loading is always something like 2-3 sec long and I have to reposition it which is not very precise.

Have you any idea how I can workaround this issue ?

Thanks,

Antoine

jie326513988, 2024-08-31:
Can two audio streams be played simultaneously, which is equivalent to mixing?

AntoineLep, 2024-09-03:
Well, here I don't want to mix 2 audios.
I just want to pause audio1 and then play audio2
And then playback audio1 when audio2 is finished

closed 2024-10-18 by github-actions[bot]