schreibfaul1/ESP32-audioI2S #825: Play streaming audio.connecttohost()  with authorization: bearer

VaAndCob, 2024-09-04:
How can I play streaming with request header "Authorization: Bearer {accesstoken}"  ?

VaAndCob, 2024-09-05:
> Nobody has asked about this yet. I can add that, but I don't have the opportunity to test it.

Firstly, I have to thank you a lot for this library, it makes thing a lot easier to many application.
this is what I added to audio.h and audio.ccp

`// LINE voice message
bool Audio::lineVoiceMessage(String messageId, const char* token) {

  String host_chr = "https://api-data.line.me/v2/bot/message/" + messageId + "/content";
    
    // Now you can convert 'host' to a C-string if needed
    const char* host = host_chr.c_str();
     xSemaphoreTake(mutex_playAudioData, portMAX_DELAY);

    if(host == NULL) {
        AUDIO_INFO("Hostaddress is empty");
        stopSong();
        xSemaphoreGive(mutex_playAudioData);
        return false;
    }

    uint16_t lenHost = strlen(host);

    if(lenHost >= 512 - 10) {
        AUDIO_INFO("Hostaddress is too long");
        stopSong();
        xSemaphoreGive(mutex_playAudioData);
        return false;
    }

    /**/
    int   idx = indexOf(host, "http");
    char* l_host = (char*)malloc(lenHost + 10);
    if(idx < 0) {
        strcpy(l_host, "http://");
        strcat(l_host, host);
    }                                      // amend "http;//" if not found
    else { strcpy(l_host, (host + idx)); } // trim left if necessary
  
    char* h_host = NULL; // pointer of l_host without http:// or https://
    if(startsWith(l_host, "https")) h_host = strdup(l_host + 8);
    else h_host = strdup(l_host + 7);

    // initializationsequence
    int16_t  pos_slash;     // position of "/" in hostname
    int16_t  pos_colon;     // position of ":" in hostname
    int16_t  pos_ampersand; // position of "&" in hostname
    uint16_t port = 80;     // port number

    // In the URL there may be an extension, like noisefm.ru:8000/play.m3u&t=.m3u
    pos_slash = indexOf(h_host, "/", 0);
    pos_colon = indexOf(h_host, ":", 0);
    if(isalpha(h_host[pos_colon + 1])) pos_colon = -1; // no portnumber follows
    pos_ampersand = indexOf(h_host, "&", 0);

    char* hostwoext = NULL; // "skonto.ls.lv:8002" in "skonto.ls.lv:8002/mp3"
    char* extension = NULL; // "/mp3" in "skonto.ls.lv:8002/mp3"

    if(pos_slash > 1) {
        hostwoext = (char*)malloc(pos_slash + 1);
        memcpy(hostwoext, h_host, pos_slash);
        hostwoext[pos_slash] = '\0';
        uint16_t extLen = urlencode_expected_len(h_host + pos_slash);
        extension = (char*)malloc(extLen + 20);
        memcpy(extension, h_host + pos_slash, extLen);
        urlencode(extension, extLen, true);
    }
    else { // url has no extension
        hostwoext = strdup(h_host);
        extension = strdup("/");
[... 100 lines cut ...]
        extension = NULL;
    }
    if(l_host) {
        free(l_host);
        l_host = NULL;
    }
    if(h_host) {
        free(h_host);
        h_host = NULL;
    }
    xSemaphoreGive(mutex_playAudioData);
    return res;
 
    
}`
and this is the result, I think it works but the download audio file (.m4a) don't play. 

_[audio.zip](https://github.com/user-attachments/files/16882878/audio.zip)_

**I have download this audio.m4a file to my PC and it can be played with music player, 
I also tried save this file to sd card and play it with sd card,  still not play.**

I'm not sure if the audio file has missing info , or etc..  
I also attached example m4a file in case you will be kind to test.
Thank you.

schreibfaul1, 2024-09-06:
The audio.m4a file is within the specification, but differs from the "usual" m4a files when analysing the m4a header.
I have extended the routine for reading the m4a header. The file can now be played.

VaAndCob, 2024-09-06:
> The audio.m4a file is within the specification, but differs from the "usual" m4a files when analysing the m4a header. I have extended the routine for reading the m4a header. The file can now be play
Thank you so much,  it works for now..  I download m4a file to littleFS and playback. 
but the problem is the speed is slower than usual. it's like 0.5 time speed or lower

Downloaded 1024 bytes...
Downloaded 1024 bytes...
Downloaded 1024 bytes...
Downloaded 1024 bytes...
Downloaded 534 bytes...
File '/voicemessage.m4a' saved, size: 9750 bytes
Playing the audio file...
info        buffers freed, free Heap: 93868 bytes
info        Reading file: "/voicemessage.m4a"
info        AACDecoder has been initialized, free Heap: 92952 bytes , free stack 2908 DWORDs
info        ch; 1, bps: 16, sr: 16000
info        AudioType: MPEG4 / Audio
info        max bitrate: 16384
info        avg bitrate: 16384
info        AudioObjectType: AAC Low Complexity
info        Sampling Frequency: 16000
info        Channel Configurations: front-center
info        AAC FrameLength: 1024 bytes
info        ch; 1, bps: 16, sr: 16000
info        Audio-Length: 8705
info        Content-Length: 9750
info        stream ready
info        syncword found at pos 0
info        Channels: 1
info        SampleRate: 16000
info        BitsPerSample: 16
info        BitRate: 16375
info        Closing audio file "voicemessage.m4a"
voicemessage.m4a
info        End of file "voicemessage.m4a"

VaAndCob, 2024-09-07:
oh and if possible you can add a function to play stream from https get with authorization: bearer would be very appreaciate

schreibfaul1, 2024-09-07:
Can you please attach the file, the last one was too short to recognise. Maybe there is a problem with mono? 
I don't want to write a separate function for the "bearer", maybe we can send the token as a user in connecttohost()?

VaAndCob, 2024-09-08:
> Can you please attach the file, the last one was too short to recognise. Maybe there is a problem with mono? I don't want to write a separate function for the "bearer", maybe we can send the token as a user in connecttohost()?

that would be great, 
the m4a file is Stereo, Sampling rate 32000Hz, 32 bps
here is an example voice sound I downloaded from Line application.  whenever I download from Line server and save to LittleFS and m4a file extension, then play with audio.connecttoFS function.  the sound volume is low and speed is slow, (sound like monster LOL)

[voice_703585.m4a.zip](https://github.com/user-attachments/files/16920580/voice_703585.m4a.zip)

VaAndCob, 2024-09-08:
one more thing,  I notice the m4a information from success play file vs no sound file as below.
SUCCESS PLAYED

File '/voicemessage.m4a' saved, size: 11626 bytes
Playing the audio file...
info        PSRAM found, inputBufferSize: 638965 bytes
info        buffers freed, free Heap: 94252 bytes
info        Reading file: "/voicemessage.m4a"
info        AACDecoder has been initialized, free Heap: 93336 bytes , free stack 3004 DWORDs
info        ch; 2, bps: 16, sr: 16000
info        AudioType: MPEG4 / Audio
info        max bitrate: 16295
info        avg bitrate: 16295
info        AudioObjectType: AAC Low Complexity
info        Sampling Frequency: 16000
info        Channel Configurations: front-center
info        AAC FrameLength: 1024 bytes
info        ch; 2, bps: 16, sr: 16000
info        Audio-Length: 10506
info        Content-Length: 11634
info        stream ready
info        syncword found at pos 0
info        Channels: 1
info        SampleRate: 16000
info        BitsPerSample: 16
info        BitRate: 23875
info        Closing audio file "voicemessage.m4a"
voicemessage.m4a
info        End of file "voicemessage.m4a"

//----------------------
NO SOUND

File '/voicemessage.m4a' saved, size: 16007 bytes
Playing the audio file...
info        buffers freed, free Heap: 89492 bytes
info        Reading file: "/voicemessage.m4a"
info        AACDecoder has been initialized, free Heap: 88576 bytes , free stack 3004 DWORDs
info        ch; 1, bps: 16, sr: 16000
info        AudioType: MPEG4 / Audio
info        max bitrate: 16384
info        avg bitrate: 16384
info        AudioObjectType: AAC Low Complexity
info        Sampling Frequency: 16000
info        Channel Configurations: front-center
info        AAC FrameLength: 1024 bytes
info        ch; 1, bps: 16, sr: 16000

//----------------
So I guessed it the m4a file itself that I downloaded from Line server that might be missing information to get the encoder play it correctly..

closed 2024-10-23 by github-actions[bot]