schreibfaul1/ESP32-audioI2S #827: Possible invalid char* pointer in connecttoFS()

anp59, 2024-09-04:
Hi,
during a test I realised that files without file extension lead to an invalid char* pointer in AUDIO_INFO, because the variable dotPos has the value -1 in this case.
```
uint8_t dotPos = lastIndexOf(afn, ‘.’);
for(uint8_t i = dotPos + 1; i < strlen(afn); i++) { afn[i] = toLowerCase(afn[i]); }

 if(endsWith(afn, ‘.mp3’)) m_codec = CODEC_MP3; // m_codec is by default CODEC_NONE
 ...
 if(m_codec == CODEC_NONE) AUDIO_INFO(‘The %s format is not supported’, afn + dotPos);
```

Best regards 
Andreas

schreibfaul1, 2024-09-05:
Hello Andreas,
you are absolutely right. If the file has no extension, -1 becomes 255 in uint8_t.
This is not intended,

best regards
Wolle

anp59, 2024-09-05:
Hi Wolle, unfortunately the fix does not yet solve the problem. The invalid pointer afn+**(-1)** is still passed to audio_info(const char *info) as *info. 
Without a file extension, the file extension check should still be excluded.

best regards 
Andreas

closed 2024-10-20 by github-actions[bot]