schreibfaul1/ESP32-audioI2S #828: slow stream, dropouts are possible

lidense, 2024-09-04:
I have made a webradio which works well with this library for most stations using my wifi. But if I am on the move and want to use a wifi hotspot using my 4G phone I get "slow stream, dropouts are possible". 
I tried the wifihighthruput version which uses Arduino as a component but does not work very well (see previous posts).
Any other solution?

schreibfaul1, 2024-09-04:
In any case, no improvement is possible with the pre-compiled Arduino version. I have quite good experience with Arduino as a component. But I am not a WiFi/LWIP expert, the settings in menuconfig may not be optimal. But you have the option of testing your own settings.

lidense, 2024-09-04:
OK thanks for your feedback.

closed 2024-10-19 by github-actions[bot]

schreibfaul1, 2025-08-13:
You can start by sending a ping to the host. If the response time is > 100 ms, it will be difficult.
Otherwise, you can use this template, which the sdkconfig is already optimised.
https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF