schreibfaul1/ESP32-audioI2S #833: Tonmischung

spikepavel, 2024-09-09:
Guten Tag. Ich versuche, während der Wiedergabe eines Tracks einen kurzen Ton (Mix) in den Audiopuffer zu mischen. Bitte sagen Sie mir, wie Sie eine ähnliche Funktion zu Ihrer Bibliothek hinzufügen können. Ich nehme den aktuellen Audiopuffer, füge ihn mit dem zusätzlichen Ton hinzu und durchschnitte die Werte...aber am Ende kommt es zum Absturz.

schreibfaul1, 2024-09-09:
Hallo,
das wird nicht einfach werden,
Du kannst audio_process_i2s() dafür verwenden. Das ist wie bei den anderen Events z.B. audio_info(const char *info)
Die Funktion ist eigentlich dazu da, das Signal auszukoppeln und extern weiterzuverwenden. Aber du kannst das Audiosignal auch manipulieren. Deine externe Audioquelle sollte die gleiche Abtastrate wie der Stream haben.
Bei diesem Beispiel wird dem Signal ein leiser Sinuston beigemischt.
`````c++
void audio_process_i2s(int16_t* outBuff, uint16_t validSamples, uint8_t bitsPerSample, uint8_t channels, bool *continueI2S){

    int16_t sineWaveTable[44] = {
         0,   3743,   7377,  10793,  14082,  17136,  19848,  22113,  23825,  24908,
      25311,  24908,  23825,  22113,  19848,  17136,  14082,  10793,   7377,   3743,
         0,  -3743,  -7377, -10793, -14082, -17136, -19848, -22113, -23825, -24908,
     -25311, -24908, -23825, -22113, -19848, -17136, -14082, -10793,  -7377,  -3743
    };

    static uint8_t tabPtr = 0;
    int16_t* sample[2]; // assume 2 channels, 16bit
    for(int i= 0; i < validSamples; i++){
        *(sample + 0) = outBuff + i * 2;     // channel left
        *(sample + 1) = outBuff + i * 2 + 1; // channel right

        *(*sample + 0) = (sineWaveTable[tabPtr] /50 + *(*sample + 0));
        *(*sample + 1) = (sineWaveTable[tabPtr] /50 + *(*sample + 1));
        tabPtr++;
        if(tabPtr == 44) tabPtr = 0;
    }
   *continueI2S = true;
}

spikepavel, 2024-09-09:
Vielen Dank. Ich mache einen Computer für Kinder, der Programmierunterricht gibt. Darauf werden die Kinder programmieren lernen. Wir nutzen Ihre Bibliothek aktiv. Ich stelle Sie als Beispiel für eine Person dar, die Menschen hilft und zeigt, wofür Programmierung notwendig ist. Danke nochmal!

spikepavel, 2024-09-09:
Еще раз здравствуйте. Добавьте это в примеры использования. Думаю это многим будет полезно. Спасибо Вам еще раз!
Hallo nochmal. Fügen Sie dies den Anwendungsbeispielen hinzu. Ich denke, das wird vielen nützlich sein. Danke nochmal!

closed 2024-11-04 by github-actions[bot]