schreibfaul1/ESP32-audioI2S #837: what causes this? E (48191) i2s_common: i2s_channel_write(1110): this channel is not tx channel

devpras22, 2024-09-11:
I get this error from time to time and the esp32 restarts. its it a buffer issue? wifi connection issue? anyone else facing this. is there a fix coming up?  here are the logs - 

Connecting WLAN .. Done, device connected.
SD Card initialized.
Button pressed. Start recording...
Connected to Backend
Recording finished. Now streaming audio chunks to backend...
Sending end-of-stream flag.
Audio data stream ended. Waiting for response...
Connecting to backend to receive GPT response...
Connected to backend, requesting GPT response...
OpenAI speaking: {"gptResponse":"The moon is about 238,855 miles away from Earth. That's really far!"}
Battery Voltage: 3.82
Battery Voltage: 3.82
Battery Voltage: 3.91
Battery Voltage: 3.91
Button pressed. Start recording...
Connected to Backend
Recording finished. Now streaming audio chunks to backend...
Sending end-of-stream flag.
Audio data stream ended. Waiting for response...
Connecting to backend to receive GPT response...
Connected to backend, requesting GPT response...
OpenAI speaking: {"gptResponse":"Stars twinkle because their light passes through the Earth's atmosphere. The air is always moving and full of different temperatures, which makes the light shift and change. This is what makes stars look like they are sparkling!"}
E (48191) i2s_common: i2s_channel_write(1110): this channel is not tx channel
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40081468  PS      : 0x00060033  A0      : 0x80083d84  A1      : 0x3ffc062c  
A2      : 0x3ffbcfbc  A3      : 0x00000000  A4      : 0x3ffa7fbe  A5      : 0x00060023  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x00000000  A9      : 0x3ffc061c  
A10     : 0x00000001  A11     : 0x0000007a  A12     : 0xb33fffff  A13     : 0x3f42ff18  
A14     : 0x00000003  A15     : 0x00060923  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40081465:0x3ffc062c |<-CORRUPTED

ELF file SHA256: 3a376e63c058246c

Rebooting...

schreibfaul1, 2024-09-13:
That's strange, can you reproduce the error. Unfortunately I don't have an OpenAI subscription.

devpras22, 2024-09-14:
I am experiencing an I2S TX channel error during audio playback. Specifically, I get the error message 'i2s_channel_write(1110): this channel is not tx channel' when trying to play audio. This error occurs inconsistently - sometimes the audio plays fine, other times it fails with this error.

The issue seems to arise when transitioning from recording audio (using I2S for input) to playing audio (using I2S for output).  There might be an issue with how the I2S channel is being reconfigured or managed between these two modes.

The error doesn't appear to be caused by my main program flow, as I use flags to ensure clear separation between recording and playback phases. Instead, it seems to be related to how the Audio library is managing the I2S channel internally, particularly when switching from input to output mode.
 
I have tried to structure my code as close to the kalo project - where i dont get this error at all. the biggest difference in my proejct and kalo is that I stream my audio directly while recording without saving on the sd card first. recording and streaming this audio out has no issues whatsoever. but something during playback, it crashes. sometimes midway during the playback and sometimes right when the playback starts. what's even weird is on standby, if my battery is depleting i have set a function where it says battery low please charge me - it sometimes crashes then even. so idle state - no recording audio prior to this and then playback still fails. 

Hope this explanation helps.

devpras22, 2024-09-16:
> That's strange, can you reproduce the error. Unfortunately I don't have an OpenAI subscription.

SO i did a test by playing from the sd card and here's the difference between playing a wav file and the mp3 file. the wav file is 7 seconds long by the way and all i hear is a chirp noise for less than a second. here are the logs 

for the wav file  - 
13:29:25.309 -> rst:0x1 (POWERON_RESET),boot:0x13 (SConnecting WLAN .. Done, device connected.
13:29:26.902 -> SD Card initialized successfully
13:29:27.000 -> I2S Configuration - BCLK: 27, LRC: 26, DOUT: 25
13:29:27.000 -> Audio volume set to: 5
13:29:27.000 -> Welcome file found, attempting to play
13:29:27.033 -> Audio file size: 621572 bytes
13:29:27.132 -> Audio config - channels: 2, sample rate: 44100 Hz, bits per sample: 16
13:29:27.132 -> Audio playback started
13:29:27.132 -> Data Rate: 0.00 kbps
13:29:27.727 -> Time: 1 ms, Bytes read: 59053/621572, Running: Yes
13:29:28.157 -> Data Rate: 800.10 kbps
13:29:28.719 -> Time: 1 ms, Bytes read: 149165/621572, Running: Yes
13:29:28.719 -> Time: 2 ms, Bytes read: 149165/621572, Running: Yes
13:29:29.148 -> Data Rate: 704.51 kbps
13:29:29.742 -> Time: 2 ms, Bytes read: 239277/621572, Running: Yes
13:29:29.774 -> Time: 3 ms, Bytes read: 241325/621572, Running: Yes
13:29:30.171 -> Data Rate: 688.13 kbps
13:29:30.765 -> Time: 3 ms, Bytes read: 331437/621572, Running: Yes
13:29:30.834 -> Time: 4 ms, Bytes read: 333485/621572, Running: Yes
13:29:31.164 -> Data Rate: 720.90 kbps
13:29:31.825 -> Time: 4 ms, Bytes read: 421549/621572, Running: Yes
13:29:31.825 -> Time: 5 ms, Bytes read: 421549/621572, Running: Yes
13:29:32.158 -> Data Rate: 688.13 kbps
13:29:32.815 -> Time: 5 ms, Bytes read: 509613/621572, Running: Yes
13:29:32.847 -> Time: 6 ms, Bytes read: 513709/621572, Running: Yes
13:29:33.149 -> Data Rate: 720.90 kbps
13:29:33.844 -> Time: 6 ms, Bytes read: 601773/621572, Running: Yes
13:29:33.875 -> Time: 7 ms, Bytes read: 603821/621572, Running: Yes
13:29:34.072 -> Reached end of file.
13:29:34.072 -> Final playback time: 7 ms, Total bytes read: 621572
13:29:34.072 -> Playback ended. Final time: 7 ms, Final bytes: 621572/621572, Running: Yes
13:29:34.072 -> Total playback duration: 193 ms
13:29:34.072 -> Playback completed successfully.

for the mp3 file - 

13:39:09.285 -> rst:0x1 (POWERON_RESET),boot:0x13 (SConnecting WLAN .. Done, device connected.
13:39:10.977 -> SD Card initialized successfully
13:39:11.075 -> I2S Configuration - BCLK: 27, LRC: 26, DOUT: 25
13:39:11.075 -> Audio volume set to: 5
13:39:11.075 -> Welcome file found, attempting to play
13:39:11.108 -> Audio file size: 246240 bytes
13:39:11.207 -> Audio config - channels: 2, sample rate: 44100 Hz, bits per sample: 16
13:39:11.239 -> Audio playback started
13:39:11.239 -> Data Rate: 0.00 kbps
13:39:11.404 -> E (5775) i2s_common: i2s_channel_write(1110): this channel is not tx channel
13:39:11.404 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

schreibfaul1, 2024-09-16:
I got myself a new openAI key, result:
[image]

and ChatGPT says:
No, the OpenAI voice service (like GPT-4) is generally not available for free to non-subscribers. OpenAI offers paid plans that allow users to access their API services, including the language models such as GPT-3 and GPT-4. However, there are some free usage options: Free quota: OpenAI often offers new users a starting credit to try out the API for a limited time.

So: I can't test anything

schreibfaul1, 2024-09-16:
I would need a short piece of the mp3 or wav file