schreibfaul1/ESP32-audioI2S #838: different speed playing mp3

ClaudioFoscan, 2024-09-13:
hi 

i got a exp32-s3 setup with a max98357a playing mp3s from a sd-card.
i updatet some of the librarys, inluding your audioI2S lib, recompiled and testet the player. 
i got some strange behavier:

mp3 with 32Bits, 48kHz Samplerate and 1Channel (Mono) and a 
mp3 with 32Bits, 48kHz Samplerate and 2Channel (Stereo)

the single Channel file plays double speed! 
was not the case with the old Librarys ...

is this a known issu? 

thanks for digging in, i'm not that good in coding to figure it out where this comes from ...

schreibfaul1, 2024-09-13:
This is not a known problem, but anything is possible, especially as there have been recent changes in the I2S area. Can you attach a small part of your mp3 file here?

ClaudioFoscan, 2024-09-16:
here the two files...
alse the printscreen from audacity
[image]
[1b.zip](https://github.com/user-attachments/files/17010178/1b.zip)

schreibfaul1, 2024-09-16:
I can't see any difference between 1.mp3 and 1b.mp3 (except 1 or 2 channels) Tested with Arduino 2.0.16 and 3.0.4 on ESP32 and ESP32-S3. Sounds like AAIICH

ClaudioFoscan, 2024-09-16:
Eis -> swissgerman for one, eins ....intressting ... is it possible that some of the esp-libs mess up ?

i'll try this evening some older esp-core libs ...

schreibfaul1, 2024-09-16:
I don't know exactly, but unfortunately I can't reproduce it.
I know that a few weeks ago (ESP-IDF V5.1) I had a bug that played mono too fast. But that has already been fixed.

devpras22, 2024-09-16:
> I can't see any difference between 1.mp3 and 1b.mp3 (except 1 or 2 channels) Tested with Arduino 2.0.16 and 3.0.4 on ESP32 and ESP32-S3. Sounds like AAIICH

which version of esp32 would you say is the most stable? i am struggling to get a successful playback on 3.0.4 with audio_play.connecttohost and audio_play.connecttospeech

ClaudioFoscan, 2024-09-17:
some additional infos: here de debug output of the two files:

(09:58:09.688) ## 2 Play Track# 0, 1/1.mp3 from 0 (34)
(09:58:09.688) info        buffers freed, free Heap: 265128 bytes (50)
(09:58:09.688) info        Reading file: "/1/1.mp3" (36)
(09:58:09.688) info        MP3Decoder has been initialized, free Heap: 260736 bytes , free stack 5296 DWORDs (93)
(09:58:09.760) info        Content-Length: 19968 (33)
(09:58:09.760) info        file has no mp3 tag, skip metadata (46)
(09:58:09.760) info        Audio-Length: 19968 (31)
(09:58:09.760) info        stream ready (24)
(09:58:09.760) info        syncword found at pos 0 (35)
(09:58:09.760) info        Channels: 1 (23)
(09:58:09.760) info        SampleRate: 48000 (29)
(09:58:09.760) info        BitsPerSample: 16 (29)
(09:58:09.760) info        BitRate: 256000 (27)
(09:58:10.092) info        Closing audio file "1.mp3" (38)
(09:58:10.092) eof_mp3     1.mp3 (17)
(09:58:10.092) info        End of file "1.mp3" (31)
(09:58:10.092) ## 3 Play Track# 1, 1/1b.mp3 from 0 (35)
(09:58:10.092) info        buffers freed, free Heap: 265128 bytes (50)
(09:58:10.092) info        Reading file: "/1/1b.mp3" (37)
(09:58:10.092) info        MP3Decoder has been initialized, free Heap: 260736 bytes , free stack 5296 DWORDs (93)
(09:58:10.092) play next Song ... (18)
(09:58:10.148) info        Content-Length: 10752 (33)
(09:58:10.148) info        file has no mp3 tag, skip metadata (46)
(09:58:10.148) info        Audio-Length: 10752 (31)
(09:58:10.148) info        stream ready (24)
(09:58:10.148) info        syncword found at pos 0 (35)
(09:58:10.148) info        Channels: 2 (23)
(09:58:10.148) info        SampleRate: 48000 (29)
(09:58:10.148) info        BitsPerSample: 16 (29)
(09:58:10.148) info        BitRate: 128000 (27)
(09:58:10.789) info        Closing audio file "1b.mp3" (39)
(09:58:10.789) eof_mp3     1b.mp3 (18)
(09:58:10.845) info        End of file "1b.mp3" (32)

Whats really interesting is: the audio-length: for the 1.mp3 (fast playing sample with one channel) its evel longer as the audio-length for 1b.mp3 (the correct playing example wiht two channels)...

ClaudioFoscan, 2024-09-17:
i just testet the following combination:

esp32-audioI2C_2.0.6 with esp32 2.0.17 -> works fine as it should, playes both files the same speed ...
esp32-audioI2C_2.0.6 with esp32 3.0.4 -> compiler error ...
esp32-audioI2C_3.0.0 with esp32 3.0.4 -> just playing once with jitter ...
esp32-audioI2C_3.0.7 with esp32 3.0.4 -> compiler error ...
esp32-audioI2C_3.0.8 with esp32 3.0.4 ->  works fine as it should ...
esp32-audioI2C_3.0.12 with esp32 3.0.4 -> works fine as it should ...

whats interesting: i could not reporduce the wrong play-speed ... 
i only changed these two librarys ... naybe i used a different esp32 lib version. 

i'll stick to 3.0.12 with 3.0.4 for now ...

closed 2024-11-01 by github-actions[bot]