schreibfaul1/ESP32-audioI2S #840: Compilation errors in VSCode/Platformio against seeed_xiao_esp32s3

vladkozlov69, 2024-09-14:
Hello,

I'm getting some compilation issues.

Here is the platformio.ini:

```
[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
lib_deps =
    https://github.com/schreibfaul1/ESP32-audioI2S.git
    smougenot/TM1637 @ 0.0.0-alpha+sha.9486982048
    mathieucarbou/AsyncTCP@^3.2.4
    mathieucarbou/ESPAsyncWebServer@^3.1.2  
    ayushsharma82/ElegantOTA @ ^3.1.4
    knolleary/PubSubClient@^2.8
build_flags = 
	-DELEGANTOTA_USE_ASYNC_WEBSERVER=1
	-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO
	-DENABLE_BUTTONS
```

and here are the errors:

```
Compiling .pio/build/seeed_xiao_esp32s3/lib8bb/ESP32-audioI2S/aac_decoder/libfaad/neaacdec.c.o
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.c: In function 'getdword_n':
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.c:1681:51: error: expected expression before '[' token
         ((uint8_t*)&tmp)[1] = ((uint8_t*)mem)[2]; [[fallthrough]];
                                                   ^
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.c:1681:53: error: 'fallthrough' undeclared (first use in this function)
         ((uint8_t*)&tmp)[1] = ((uint8_t*)mem)[2]; [[fallthrough]];
                                                     ^~~~~~~~~~~
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.c:1681:53: note: each undeclared identifier is reported only once for each function it appears in
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.c:1683:51: error: expected expression before '[' token
         ((uint8_t*)&tmp)[2] = ((uint8_t*)mem)[1]; [[fallthrough]];
                                                   ^
*** [.pio/build/seeed_xiao_esp32s3/lib8bb/ESP32-audioI2S/aac_decoder/libfaad/neaacdec.c.o] Error 1
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'int AACDecode(uint8_t*, int32_t*, short int*)':
.pio/libdeps/seeed_xiao_esp32s3/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:142:63: error: invalid conversion from 'uint32_t*' {aka 'unsigned int*'} to 'long unsigned int*' [-fpermissive]
             int8_t err = NeAACDecInit2(hAac, specificInfo, 2, &aacSamplerate, &aacChannels);(void)err;
```

Replacing 

```
uint32_t aacSamplerate = 0;
```
to

```
unsigned long aacSamplerate = 0;
```

in aac_decoder.cpp

and removing "[[fallthrough]]" in neaacdec.c helps to fix it.

schreibfaul1, 2024-09-14:
Thanks for your contribution. I have recently added floating point arithmetic to the aac decoder. In doing so, I included the "[[fallthrough]]". This has been around since C++17. You are probably using a different compiler. I hope the error is gone now.

vladkozlov69, 2024-09-15:
Just figured out that you use different platform/packages in your radio example:

```
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.4
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.4/esp32-arduino-libs-3.0.4.zip
```
So I applied these changes to my code and it compiles fine now. Thanks!

closed 2024-09-15 by vladkozlov69