schreibfaul1/ESP32-audioI2S #841: Use "connecttoFS()" with USB Flash/Thumb Drive on ESP32s3

BoltSwith, 2024-09-14:
Greetings,

I've implemented this library in my project and with works great.
I use the esp32s3 module due to it's advance capabilities especially it's USB Host features. The project is designed to use USB thumb/flash drive as mass storage device. I use the ESP-IDF v5.1.3 with VS Code as editor. I added Arduino as a component (for processing/interfacing arduino related libs).

I am able to read and write usb flash drives using this lib - [USB Host MSC (Mass Storage Class) Driver](https://github.com/espressif/esp-usb/tree/master/host/class/msc/usb_host_msc) provided in the official Espressif Component Registry.

Assessing the ESP32-audioI2S, I can see that it uses the Arduino style File System Access/Functions, however the USB Host MSC library mentioned above uses the standard C-style File System Access/Functions. To enable me use this library to process audio files on the USB thumb drive, I attempted creating an Arduino FS wrapper class (similar to SD_MMC, SPIFFS or LittleFS classes/objects), which wraps the C-Style functions provided by the USB Host MSC lib such as USBMsc.begin() is used to initialize the USB interface and mount the drive etc. **All my attempts and refactoring of the wrapper class (in the last 2 weeks) to work with the ESP32-audioI2S lib via the connecttoFS() function failed.**

I wish to use this ESP32-audioI2S library to process audio files in a USB Flash drive. Seems there is no official Arduino library that reads/write USB flash drives on an esp32 (particularly esp32/s3/c3/s2).

**Is it possible to create an overloaded "connecttoFS()" function dedicated to processing audio files on usb flash/thumb drives using the standard C-style File System operations/functions?** I believe this is an easier route to go than creating an Arduino FS wrapper Class.

Espressif provided an example on mounting and carrying out FS operations on a USB flash drive in this example - [msc_host_example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/msc)

schreibfaul1, 2024-09-15:
The problem is not so easy to solve. It would be nice if external storage devices with access via USB were integrated into the file system. That would be a task for the Espressif team. In the case of in-house development, the standard functions provided by FFAT, SPIFFS... (position, seek, available, read ...) It will not be enough just to change something in connecttoFS().
[msc_host_example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/msc) can be a good start.

BoltSwith, 2024-09-15:
Thanks for the swift response. 

Hopefully, someday soon, the Espressif Team implements and integrate a USB FS Class to perform Arduino Style File System processes.

closed 2024-09-15 by BoltSwith

schreibfaul1, 2024-09-15:
I see you have developed an impressive project. 
https://github.com/BoltSwith/ESP32_CD_Player

Then you will also be able to expand the FS.

[SSD.zip](https://github.com/user-attachments/files/17007027/SSD.zip)
(You can use SPIFFS as a template)
````c++
#include "Arduino.h"
#include "WiFi.h"
#include "SSD.h"
File = SSD. ...... // your own impl.

BoltSwith, 2024-09-18:
I've attempted this before i.e. expanding the FS. I just couldn't get it to work. At this point, I'll wait until an official library (UsbFS.h and UsbFS.cpp) is developed by someone who knows the Arduino FS implementation in-depth. Thanks.