schreibfaul1/ESP32-audioI2S #845: no playback: format of incoming mp3 stream (lame 3.1) doesn't match what esp32 detects

devpras22, 2024-09-18:
I am unable to play the streamed audio from my backend server to the esp32. here are the logs of the esp32 - you can see for some reason it is detecting an mpeg version 2 later 3 file as mpeg version1 layer 1. Do i need to convert the mp3 file? I am so clueless why this happening to me. it's not a hardware issue as kalo project works fine.  I have even tried testing by downgrading the esp board. 

LOGS FROM ESP32 - 13:33:31.480 -> �=��Connecting to host and starting playback...
13:33:39.977 -> info        PSRAM not found, inputBufferSize: 13951 bytes
13:33:39.977 -> info        buffers freed, free Heap: 178084 bytes
13:33:39.977 -> info        connect to: "b5f20fa4-70d9-43bf-84e4-7e1a5c8d9f4d-00-25svxhvj7i2k2.sisko.replit.dev" on port 443 path "/audio/stream"
13:33:40.276 -> info        SSL has been established in 303 ms, free Heap: 137160 bytes
13:33:41.930 -> info        MP3Decoder has been initialized, free Heap: 109100 bytes , free stack 1968 DWORDs
13:33:41.930 -> lasthost    https://b5f20fa4-70d9-43bf-84e4-7e1a5c8d9f4d-00-25svxhvj7i2k2.sisko.replit.dev/audio/stream
13:33:41.930 -> Time: 0.00 s, Bytes read: 0, Running: Yes
13:33:41.930 -> Data Rate: 0.00 kbps
13:33:41.962 -> info        file has no mp3 tag, skip metadata
13:33:41.962 -> info        Audio-Length: 41280
13:33:41.962 -> info        Webfile: stream ready, buffer filled in 15 ms
13:33:41.962 -> info        syncword found at pos 0
13:33:41.962 -> info        MPEG-1, Layer I
13:33:41.962 -> info        Channels: 1
13:33:41.962 -> info        SampleRate: 24000
13:33:41.962 -> info        BitsPerSample: 16
13:33:41.962 -> info        BitRate: 160000
13:33:42.948 -> Time: 0.00 s, Bytes read: 0, Running: Yes
13:33:42.948 -> Data Rate: 0.00 kbps
13:33:43.936 -> Time: 0.00 s, Bytes read: 0, Running: Yes
13:33:43.936 -> Data Rate: 0.00 kbps
13:33:44.034 -> info        End of webstream: "https://b5f20fa4-70d9-43bf-84e4-7e1a5c8d9f4d-00-25svxhvj7i2k2.sisko.replit.dev/audio/stream"
13:33:44.034 -> Playback ended. Final time: 1 ms, Final bytes: 0, Running: No
13:33:44.034 -> Playback ended. Final time (in seconds): 0.00 s 

This is the file info - General
Complete name                            : audio_1726646621752.mp3
Format                                   : MPEG Audio
File size                                : 40.3 KiB
Duration                                 : 2 s 64 ms
Overall bit rate mode                    : Constant
Overall bit rate                         : 160 kb/s
Writing library                          : LAME3.100

Audio
Format                                   : MPEG Audio
Format version                           : Version 2
Format profile                           : Layer 3
Duration                                 : 2 s 64 ms
Bit rate mode                            : Constant
Bit rate                                 : 160 kb/s
Channel(s)                               : 1 channel
Sampling rate                            : 24.0 kHz
Frame rate                               : 41.667 FPS (576 SPF)
Compression mode                         : Lossy
Stream size                              : 40.3 KiB (100%) / 40.3 KiB (100%)
Writing library                          : LAME3.100

closed 2024-09-18 by devpras22

channel11, 2024-11-05:
How did you solve this? I try to play an ffmpeg stream, with similar result. No sound at all.

devpras22, 2024-11-05:
i had conflicting code during audio.play.. cleaned it up and it started working. sorry i cant be super specific my codebase has changed drastically in the last 2 months so i cant pin point the exact code but there was some if statements that didn't trigger the audio loop correctly.