schreibfaul1/ESP32-audioI2S #847: Can't play an MP3 file from an SD card more than once.

kidmirage, 2024-09-20:
I want to play multiple MP3 files in order much like your playlist example. Unfortunately only the first selected file plays. A simple example.

`// Include required libraries
#include "Arduino.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// microSD Card Reader connections
#define SD_CS          5
#define SPI_MOSI      23 
#define SPI_MISO      19
#define SPI_SCK       18
 
// I2S Connections
#define I2S_DOUT      22
#define I2S_BCLK      26
#define I2S_LRC       25

// Create Audio object
Audio audio;

void setup() {
    
    // Set microSD Card CS as OUTPUT and set HIGH
    pinMode(SD_CS, OUTPUT);      
    digitalWrite(SD_CS, HIGH); 
    
    // Initialize SPI bus for microSD Card
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    
    // Start Serial Port
    Serial.begin(115200);
    Serial.println("Starting.");
    Serial.println("");
    
    // Start microSD Card
    if(!SD.begin(SD_CS))
    {
      Serial.println("Error accessing microSD card!");
      while(true); 
    }
    
    // Setup I2S 
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    
    // Set Volume
    audio.setVolume(18);
    
    // Open music file
    audio.connecttoFS(SD,"/Trekulator_Sound.mp3");
    // audio.setFileLoop(true);

    // Only one speaker.
    audio.forceMono(true);
}
 
void loop(){
  audio.loop();
}

void audio_eof_mp3(const char *info){  //end of file

  Serial.print("+");
  audio.connecttoFS(SD,"/Scanner_Sound.mp3");
}`

Only Trekulator_Sound.mp3 plays. I can see that audio_eof_mp3 gets executed but /Scanner_Sound.mp3 does not play.

morgan-wild, 2024-09-21:
Same issue for me here.

I moved back to an old version (that is not easy to give its number because the value in the library.properties file looks stucked to 2.0.0).

This old version stay uses connecttoSD(path) instead of connecttoFS(SD, path) and works perfectly with successive readings.

```
        // Latest version,reads the first file but fires a 'processLocalFile(): audioHeader reading timeout' for next ones
        // _audio.connecttoFS(SD, trackPath.c_str());

        // Initial version of the project ('lib'): Works well
        _audio.connecttoSD(trackPath.c_str());
```

I also play and play another file when audio_eof_mp3 is called.

Maybe something is broken?

schreibfaul1, 2024-09-22:
This seems to have been a timing problem. It did not affect older versions. Hope it works again.

kidmirage, 2024-09-22:
That seems to have fixed things on my end. Thank you.

morgan-wild, 2024-09-23:
Fixed on my side too! Great job!

ra0943-VK3ACH, 2024-11-13:
This doesnt work for me either does anyone have a solution?