schreibfaul1/ESP32-audioI2S #849: No function to output audio to one GPIO pin

moadib-programmer, 2024-09-21:
I want to map output to only GPIO pin which will connected to the headset, setPin function takes arguments but onoy need to set the outpin

schreibfaul1, 2024-09-22:
The ESP32, (not ESP32-S3) has a built-in DAC on pin 25 and 26. Espressif supports the DAC up to Arduino version 2.0.17. However, the quality is low because of the 8 bits per sample.

moadib-programmer, 2024-09-22:
Hi @schreibfaul1  thank you for replying but i guess i didnt make my question clear.

I wan to get the output of audio one pin instead of these three, which can be directly hooked up to speaker's positive terminal

schreibfaul1, 2024-09-22:
You probably mean S/PDIF (Sony/ Philips Digital Interface)? This is not realised.

closed 2024-11-07 by github-actions[bot]