schreibfaul1/ESP32-audioI2S #850: connecttoFS() : Playback of small files (less than maxFrameSize) stall and do not play.

LabRat3K, 2024-09-22:
Scenario: mp3 file on SD card, with length less than maxFrameSize (debugging showed maxFrameSize was 1600, and the files were 1548, and 1440 respectively) 
Api: connecttoFS(SD, <filename>) 

Result: Files fail to playback, logs stop at the "MP3Decoder has been initialized..." statement. 
Expected Result: Audio would play until EOF.

----
Fix: (apologies that the machine I'm on doesn't have my gitlab ssh credentials, so can't setup a pull request just now)
Add an additional condition to determine if/when to pass read data to the readAudioHeader parsing.  Suppose could also be solved by ensuring that maxFrameSize never is exceeds the overall size of the file, but this was the fix that got my audio playing. 

Change
> -            if(InBuff.bufferFilled() > maxFrameSize) { // read the file header first
To
> +            if((InBuff.bufferFilled() > maxFrameSize) ||
> +               **(InBuff.bufferFilled() == m_fileSize**)) { // read the file header first
>                  InBuff.bytesWasRead(readAudioHeader(InBuff.getMaxAvailableBytes()));

schreibfaul1, 2024-09-22:
I don't think anyone has noticed this yet. Your suggestion has been realised, extremely short files from SD and from the web are now played.

LabRat3K, 2024-09-22:
wt wt. (two very short woot sounds).

closed 2024-09-22 by LabRat3K