schreibfaul1/ESP32-audioI2S #855: AAC Decoder could not be initialized

sercan45329, 2024-09-26:
Why AAC Decoder could not be initialized ?

[image]

## Code

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <Audio.h>

const char* ssid = "SSID";
const char* password = "PASSWORD";

// MAX98357 I2S pinouts
#define I2S_DOUT 27
#define I2S_BCLK 26
#define I2S_LRC  25

Audio audio;

String currentStation = "";
String currentTitle = "";
int currentBitrate = 0;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("WiFi'ye bağlanılıyor...");
  }
  Serial.println("WiFi'ye bağlandı");
  
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(10); 
  audio.forceMono(true);
  audio.connecttohost("https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.aac");
}

void loop() {
  audio.loop();
  // Diğer loop içeriği...
}

void audio_info(const char *info) {
  Serial.print("info        "); Serial.println(info);
}

```

## System Information

- ESP32 WROOM
- Arduino IDE Version: [2.3.2]
- Audio library [latest]

CeEs-atd, 2024-09-27:
Please review the controller vs codec table on the start page / readme page: 
ESP32 without PSRAM ("normal" WROOM) cannot play AAC streams due to lack of internal memory.

sercan45329, 2024-09-27:
Thank you for answering. I am so blind that I did not see the table.

closed 2024-09-27 by sercan45329