schreibfaul1/ESP32-audioI2S #856: Can you help me write the code to call Doubao TTS?

Explorerlowi, 2024-09-26:
My programming skills are so poor that I really can’t do it (灬ꈍ ꈍ灬).
Here is the relevant document: https://www.volcengine.com/docs/6561/79823
I will pay you a certain amount of compensation. Thank you very much if you can do it!

Explorerlowi, 2024-10-07:
The content returned by Doubao TTS request is not directly a binary audio stream. Its audio data is stored in the data field of a json structure. It is base64 encoded data. Binary audio data can only be obtained after base64 decoding. In this case, how to play it
[image]
[image]

schreibfaul1, 2024-10-07:
With a "normal audio stream", the data would be written to the buffer here.
InBuff.getWritePtr() is the pointer to the position from which the data is written
bytesAddedToBuffer contains the number of bytes actually written.
Then the conversion from base64 would have to be done.
[image]

You don't need to worry about the rest, if it is an MP3 stream, for example, the ID3 header is automatically loaded when the buffer is full enough and the file is played.

closed 2024-11-25 by github-actions[bot]