schreibfaul1/ESP32-audioI2S #857: NetworkClient.h: No such file or directory

serkansmg, 2024-09-29:
this is my platformio.ini

[env:esp32-s3-devkitc-1]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
framework = arduino
upload_speed = 2000000
upload_port = COM9
monitor_speed = 115200
monitor_port = COM9
board_build.flash_mode = qio
board_build.arduino.memory_type = qio_qspi
board_build.arduino.ldscript = esp32s3_out.ld
upload_resetmethod = usb_reset
board_build.arduino.psram = enabled
build_flags = 
	-DCONFIG_IDF_TARGET_ESP32S3
	-DCONFIG_SPIRAM_CACHE_WORKAROUND
	-DBOARD_HAS_PSRAM
	-DCONFIG_ESP32S3_SPIRAM_SUPPORT
	-DCONFIG_SPIRAM_SPEED_120M
	-DCONFIG_SPIRAM_SIZE=2097152
	-DCORE_DEBUG_LEVEL=3
	-mfix-esp32-psram-cache-strategy=memw
	-Wno-attributes
	-DCONFIG_HEAP_SIZE=65536
	-DCONFIG_BLUETOOTH_NIMBLE_STACK_TASK_STACK_SIZE=16384
	-DCONFIG_NIMBLE_HCI_TASK_STACK_SIZE=8192
	-DCONFIG_BT_NIMBLE_TASK_STACK_SIZE=8192
	-DCONFIG_BT_NIMBLE_CRYPTO_STACK_SIZE=8192
	-DCONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=8192
	-DCONFIG_BT_NIMBLE_CONTROLLER_TASK_STACK_SIZE=8192
	-DCONFIG_BT_NIMBLE_ENABLED=1
	-DCONFIG_BT_NIMBLE_MAX_CONNECTIONS=3
	-DCONFIG_BT_NIMBLE_MAX_BONDS=3
	-DCONFIG_BT_NIMBLE_MAX_CCCDS=8
	-DCONFIG_BT_NIMBLE_L2CAP_COC_MAX_NUM=3
	-O2
	-DCONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
	-DCONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
	-DCONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=16
	-DCONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
	-DCONFIG_LWIP_TCP_RECVMBOX_SIZE=64
	-DCONFIG_LWIP_TCP_SND_BUF_DEFAULT=8192
	-DCONFIG_LWIP_TCP_WND_DEFAULT=8192
	
	-DARDUINO_ESP32S3_DEV
	-DARDUINO_USB_CDC_ON_BOOT=0
board_build.partitions = default_8MB.csv
debug_tool = esp-builtin
debug_init_break = break setup
upload_protocol = esptool
monitor_filters = 
	colorize
	time
	esp32_exception_decoder
lib_deps = 
    https://github.com/schreibfaul1/ESP32-audioI2S.git

until 4 days ago, it was working perfectly.. i dont know what happened but now i am getting:

In file included from src/main.cpp:4:
.pio/libdeps/esp32-s3-devkitc-1/ESP32-audioI2S/src/Audio.h:20:10: fatal error: NetworkClient.h: No such file or directory

CeEs-atd, 2024-09-29:
Hi!
`NetworkClient.h` should be part of the ESP libraries - at least it is, when the ESP32 toolchain & libs are installed with the Arduino IDE: It is located in
`C:\Users\username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\Network\src`
(same path for `NetworkClientSecure.h`). 
But maybe the ethernet-PHY boards are not supported outside the Arduino Environment...?

This include-switch in `audio.h` should fix this:

```
#ifndef ETHERNET_IF
#include <WiFi.h>
#include <WiFiClientSecure.h>
#else
#include <NetworkClient.h>
#include <NetworkClientSecure.h>
#endif
```
(with the Arduino toolchain it also saves some time when building)

schreibfaul1, 2024-09-30:
NetworkClient.h and NetworkClientSecure.h have only been available since Arduino V3.
With Arduino V2, these files are no longer taken into account.

serkansmg, 2024-09-30:
> NetworkClient.h and NetworkClientSecure.h have only been available since Arduino V3. With Arduino V2, these files are no longer taken into account.

 check version 3.0.12.. the latest one on github ,which is changed in last 4-5 days have it.. 

header of audio.h in 3.0.12 :
`
#pragma once
#pragma GCC optimize ("Ofast")
#include <vector>
#include <Arduino.h>
#include <libb64/cencode.h>
#include <esp32-hal-log.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <SD.h>
#include <SD_MMC.h>
#include <SPIFFS.h>
#include <FS.h>
#include <FFat.h>
#include <atomic>

#if ESP_IDF_VERSION_MAJOR == 5
#include <driver/i2s_std.h>
#else
#include <driver/i2s.h>
#endif

#ifndef I2S_GPIO_UNUSED
  #define I2S_GPIO_UNUSED -1 // = I2S_PIN_NO_CHANGE in IDF < 5
#endif
using namespace std;
`

closed 2024-11-15 by github-actions[bot]