schreibfaul1/ESP32-audioI2S #859: Use PSRAM to store audio instead of SD card module

SleepInfinity, 2024-10-02:
Hello,

I’m currently using the esp32-audioI2S library, and I was wondering if it’s possible to store audio data in PSRAM instead of using an SD card module.

Questions:

Is there currently any way to utilize the PSRAM for this purpose?

If yes, could you provide guidance on how to implement it?

If no, is there a specific reason why this functionality hasn’t been implemented yet? Would it be possible to add it in the future?

Thank you for your great work on this library, and I appreciate any help you can provide on this topic!

schreibfaul1, 2024-10-02:
PSRAM is not good for storing files. If you have a board with 8 or 16MB flash you can set up a FAT partition or use a pre-built one like "16MB Flash(3MB APP/9.9MB FATFS)"

SleepInfinity, 2024-10-02:
oh ok,
also i wanted to use the google text to speech api with the library but didnt managed to play the audio from the i2s speaker, because the response is base64 encoded is it possible to decode it with your library ?
The curl:
```curl
curl -X POST \
  'https://texttospeech.googleapis.com/v1/text:synthesize?key=API_KEY' \
  -H 'Content-Type: application/json' \
  -d '{
    "input": {
      "text": "Hello! This is a test."
    },
    "voice": {
      "languageCode": "en-US",
      "name": "en-US-Journey-F",
      "ssmlGender": "FEMALE"
    },
    "audioConfig": {
      "audioEncoding": "LINEAR16"
    }
  }'
```

The response:
```json
{
  "audioContent": "UklGRoQEAABXQVZFZm10IBAAAAABAAEAQB8AAEAfAAABAAgAZGF0YbQEAAQ..."
}
```

schreibfaul1, 2024-10-02:
base64 encoded audio data cannot be played with the library

SleepInfinity, 2024-10-02:
> base64 encoded audio data cannot be played with the library

if i decode it with another library how can i play it ?

closed 2024-11-16 by github-actions[bot]