schreibfaul1/ESP32-audioI2S #862: [Audio.cpp:3211] processLocalFile(): audioHeader reading timeout

AntoineLep, 2024-10-08:
Hi,

I'm encountering this issue when trying to read a file.

`[Audio.cpp:3211] processLocalFile(): audioHeader reading timeout`

I have no problem reading other files and this one was readable a few releases before using this library.
I tried to increase the timeout value to 20 sec but it still fails

File is downloadable here: https://cdn.staytuned.io/misc/share/3328140023923_001.mp3

Here is my core code:

```c
#include <SD.h>
#include <Audio.h>
#include <SPI.h>
#include <Arduino.h>
#include <Wire.h>

#define SD_CS 13
#define SPI_MOSI 15
#define SPI_MISO 2
#define SPI_SCK 14
#define I2S_DOUT 26
#define I2S_BCLK 5
#define I2S_LRC 25
#define SDA 18
#define SCL 23

#define PA GPIO_NUM_21

Audio audio;

void setup() {
  Serial.begin(115200);

  Wire.begin(SDA, SCL);

  Serial.println("Configuring SPI");
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);

  // Initialize SD card
  if (!SD.begin(SD_CS)) {
    Serial.println("Failed to initialize SD card");
    return;
  }

  // Amp power enable
  gpio_reset_pin(PA);
  gpio_set_direction(PA, GPIO_MODE_OUTPUT);
  gpio_set_level(PA, 1);

  // Initialize I2S
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(21);

  // Start playing MP3 file
  audio.connecttoFS(SD, "/audiobooks/b814f4c5-40fc-4ccd-a630-f4de8f30d13e/3328140023923_002.mp3");
}

void loop() {
  if (audio.isRunning()) {
    audio.loop();
  }
}
```

Do you have any idea why it fails ?
I've seen the same issue on github but I don't have any delay()

Thanks,

Antoine

schreibfaul1, 2024-10-09:
Hello Antoine,
the error was difficult to find, especially as it rarely becomes noticeable. But it always occurs with your file. I have fixed the error, the file runs now as a web stream and as a file on the SD card.

Best regards
Wolle

AntoineLep, 2024-10-09:
Hi Wolle,

It indeed works now.

Thanks a lot

closed 2024-10-09 by AntoineLep