schreibfaul1/ESP32-audioI2S #863: Does not play local mp3 from FS (SPIFFS)

selimovur, 2024-10-09:
I try to play local and remote files in my project, it works fine when play with "connecttohost" however it does not play a simple mp3 file from spiffs even though it returns true while connecting to the file. So help pls...

Explorerlowi, 2024-10-11:
You should use connecttoFS to play mp3 files in spiffs

selimovur, 2024-10-14:
> You should use connecttoFS to play mp3 files in spiffs

Thanks for your reply, but I'm already calling it before calling audio.loop function.

selimovur, 2024-10-15:
#pragma once
#include "Audio.h"
#include <WiFi.h>

Audio audio;

#define BUTTON_PIN 27
#define SD_CS 4

const char* ssid = "??"; 
const char* password = "??";

void setup() {
  Serial.begin(115200);
  delay(1000);

  SPIFFS.begin(true);

  Serial.println("Connecting to wifi...");
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting...");
  }

  Serial.println("Connected..");
  Serial.print("IP Adress: ");
  Serial.println(WiFi.localIP());

  pinMode(BUTTON_PIN, INPUT_PULLUP);
  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);

  audio.setPinout(17, 18, 16);
  audio.setVolume(21);

  if(SPIFFS.exists("/open_sound.mp3")){
    Serial.println("File is existing");
    Serial.println(audio.connecttoFS(SPIFFS, "/open_sound.mp3"));
  }
  else {
    Serial.println("File does not exist");
  }
}

void loop() {
  audio.loop();
}

selimovur, 2024-10-15:
The file does exist and audioconnecttoFS returns 1 but not playing, when i try the same example with audioconnecttohost, it does work..

lostkk2024, 2024-10-26:
Hello, sir. I am experiencing the same issue. May I ask if you have resolved this issue?

schreibfaul1, 2024-10-26:
This library does not recognise SPIFFS, SD_MMC or FFat. These are file system objects that are transferred.
If you put a short text file into SPIFFS and then open it, is the text fully readable?

selimovur, 2024-10-26:
Yes I solved the issue, it was about the partition scheme usage due to an arduino ide bug. Once I cleared the flash memory and write the same file again with the correct partition scheme and upload a small sketch into the esp32, it started working. Thanx...

lostkk2024, 2024-10-29:
Thank you for your replay. After switching the file system to LittleFS, everything is working fine now.

Rikolleti, 2024-11-16:
> Thank you for your replies. After switching the file system to LittleFS, everything is working fine now.

Hello !
Can you please send me code for using LittleFS ?

lostkk2024, 2024-11-18:
> > Thank you for your replies. After switching the file system to LittleFS, everything is working fine now.
> 
> Hello ! Can you please send me code for using LittleFS ?

#include "Arduino.h"
#include "Audio.h"
#include "LittleFS.h"
#include "FS.h"

#define I2S_BCLK     GPIO_NUM_10
#define I2S_LRC       GPIO_NUM_44
#define I2S_DOUT     GPIO_NUM_9
#define I2S_CTRL    GPIO_NUM_5

Audio audio;

void setup() {
    Serial.begin(115200);
  
    if (!LittleFS.begin()) {
        Serial.println("Failed to initialize LittleFS");
        return;
    }
    Serial.println("LittleFS initialized");

    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);

    
    pinMode(I2S_CTRL, OUTPUT); // If this pin does not exist, these lines can be ignored.
    digitalWrite(I2S_CTRL, HIGH); // If this pin does not exist, these lines can be ignored.

    audio.setVolume(10);

    audio.connecttoFS(LittleFS, "/22.mp3");
}

void loop() {
    audio.loop();
}

closed 2025-01-07 by github-actions[bot]