schreibfaul1/ESP32-audioI2S #868: ESP32-audioI2S Internal DAC - No Audio

jolly-git, 2024-10-12:
Hi

I have tried to build https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Internal%20DAC
 But no audio output, Please help. 
#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"

Audio audio(true, 3);
WiFiMulti wifiMulti;
String ssid =     "34dd38203";
String password = "342609[61";

void setup() {
   Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
        WiFi.disconnect(true);
        wifiMulti.run();
    }
    audio.setVolume(12); // 0...21

    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3"); //  128k mp3
}

void loop()
{
    audio.loop();
}
-
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}

 I am using esp32 board package vesrion 3.0.5
[image]

jolly-git, 2024-10-12:
esp 32 driver version 

idf-release_v5.1-33fbade6

schreibfaul1, 2024-10-12:
In IDF V5 (Adruino V3...) there is no I2S support from Espressif for the internal DAC. The last working version is Arduino 2.0.17

vednildip, 2024-11-05:
[README]
ESP32-audioI2S
⚠️ This library only works on multi-core ESP32 chips like the ESP32-S3. It does not work on the ESP32-S2 or the ESP32-C3 ⚠️

closed 2024-12-22 by github-actions[bot]