schreibfaul1/ESP32-audioI2S #874: forceMono does not work.

shinoryo1216, 2024-10-21:
I tested the forceMono under different conditions. In any case, stereo sources were not mixed into mono. The meters on my mixer showed that signals from left/right channels were different. (With mono source, signals were same as expected.)

My Hardware: Esp32-Wroom-32D devkitc with 30 pins + PCM5102

I have tested only ver 3.0.12. 

It does not work even when I simply added "audio.forceMono(true);" into the original I2Saudio.ino file in the "examples" folder. Is this the way how it is supposed to work. right? Maybe I misunderstood things? I am totally an amateur with no experience of programming.

Following is the I2Saudio.ino with audio.forceMono(true).

Audio audio;
WiFiMulti wifiMulti;
String ssid =     "mywifi";
String password = "mypassword";

void setup() {
    pinMode(SD_CS, OUTPUT);
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    SPI.setFrequency(1000000);
    Serial.begin(115200);
    SD.begin(SD_CS);
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
        WiFi.disconnect(true);
        wifiMulti.run();
    }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.forceMono(true);
    audio.setVolume(20); // 0...21

//    audio.connecttoFS(SD, "test.wav");
//    audio.connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");
//    audio.connecttohost("http://somafm.com/wma128/missioncontrol.asx"); //  asx
//    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.aac"); //  128k aac
      audio.connecttohost("http://playerservices.streamtheworld.com/api/livestream-redirect/AFNP_TKO.mp3"); 
}

void loop()
{
    audio.loop();
    if(Serial.available()){ // put streamURL in serial monitor
        audio.stopSong();
        String r=Serial.readString(); r.trim();
        if(r.length()>5) audio.connecttohost(r.c_str());
        log_i("free heap=%i", ESP.getFreeHeap());
    }
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}

schreibfaul1, 2024-10-22:
Thanks for your contribution. In fact, forceMono() has been ineffective since IDF V5.... I have reactivated this function.

shinoryo1216, 2024-10-23:
I tested the modified code. Works fine with ver 3.0.12.
Thank you very much!

closed 2024-10-23 by shinoryo1216