schreibfaul1/ESP32-audioI2S #875: ESP32-audioI2S crashes on arduino-esp32 v3.0.7

enjoyneering, 2024-10-24:
Hi schreibfaul1,

The latest ESP32-audioI2S v3.0.13d crashes on arduino-esp32 v3.0.7. Same code works OK on arduino-esp32 v3.0.5. I'm not an expert, but I think the problem is related to the latest changes in PSRAM. The Exception Decoder points to Audio::setDefaults().

schreibfaul1, 2024-10-24:
The Arduino version 3.0.7 was released recently. I have tested this version in VSCode nit PIO and have not been able to reproduce the error. According to your log, there is a memory violation during an if query. Strange, maybe the error only occurs under certain configurations?

enjoyneering, 2024-10-25:
Hi schreibfaul1,

If I disable PSRAM, my code works fine on arduino-esp32 v3.0.7. Looks like problem is related to the latest changes in arduino PSRAM:
- https://github.com/espressif/arduino-esp32/pull/10406
- https://github.com/espressif/arduino-esp32/pull/10395
- https://github.com/espressif/arduino-esp32/pull/10390

enjoyneering, 2024-10-26:
I enabled PSRAM, commented out the line audio.connecttohost(AUDIO_WEBRADIO_00); and everything worked, without sound.

[image]
[image]

stef-ladefense, 2024-10-26:
hi, I have exactly the same crashes with version 3.07.

I went back to version ESP32-audioI2S-3.0.12f to work with 3.07.
with 3.05, version ESP32-audioI2S-3.0.13e works.

schreibfaul1, 2024-10-27:
I have now tested V3.0.7 in the Arduino IDE. @enjoyneering is right, PSRAM is not initialised correctly. This leads to an "out of memory".
[image]

Arduino V3.0.5 works.

me-no-dev, 2024-10-29:
The problem is not in Arduino, but in the library constructor: https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L161

It is allocating and dealing with memory BEFORE the memory is properly initialized. All that code should be executed in `setup()` and not before that.

It worked before, because it ran before any PSRAM was touched, thus acting like PSRAM is not there at all.

me-no-dev, 2024-10-29:
This does not change the fact that in the constructor (which runs before PSRAM is added to heap and anything else is started and prepared) there are many things done that should not be done there.  All that should happen in the constructor is basic class variables setup. All other memory operations should be done later in `setup()`. Starting I2S also.  That `oom` error above happened at the same time that PSRAM was enabled, which means that the code that allocates memory started even earlier. This all lands the constructor in no man's land (at the same time that memory map for the OS is being set).

me-no-dev, 2024-10-29:
This has not been a problem so far, because PSRAM was initialized AFTER `Audio::Audio()`, therefore PSRAM was not even on yet. You were starting it yourself [here](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L171) which is also a very bad idea. This is a private API that is called by the core at the appropriate time. Please revise your constructor, else coming 3.1.0 there will be even more issues with the memory

schreibfaul1, 2024-10-29:
@me-no-dev Thanks for the explanation. The PSRAM will now be initialised at a later time, everything is perfect, and in the Arduino IDE it is now also running.
I think I can close this topic with that.

closed 2024-10-29 by schreibfaul1