schreibfaul1/ESP32-audioI2S #876: Can't use a pointer to Audio class...

FatherMarco1971, 2024-10-24:
I had a project where i used define:

``Audio *audio;``
then some point in the code:
``audio = new Audio;

audio->init();

...

audio->loop();``
everything worked fine till you changed from:
`xTaskCreatePinnedToCore to xTaskCreateStaticPinnedToCore`
the started to have this error at runtime:
assert failed: xTaskCreateStaticPinnedToCore tasks.c:681 (xPortCheckValidTCBMem(pxTaskBuffer))

Changing the definition from pointer to:

``
Audio audio;
``
and so:
``audio.init();

...

audio.loop();``

everything work again.

Could someone help me to understand why (i'm not so good in C/C++...)
Thank you.

schreibfaul1, 2024-10-24:
The cause will certainly lie elsewhere. A static task has the advantage that memory fragmentation is avoided. In your case, the Task Control Block (TCB) cannot be created. The limit of the dynamic RAM has probably been reached. Can you move something to the PSRAM?

FatherMarco1971, 2024-10-28:
What's your commit for Wolle?
I don't understand...could ou help?

schreibfaul1, 2024-10-29:
You are right, something went wrong. If an object is created dynamically, the stack of the task cannot be inside the object. I have now corrected this. 

I have tested it in the Arduino IDE with Arduino version 3.0.5 
And this is my test sketch:
````c++

//********************/home/wolle/Arduino/audioI2S_ptr/src/Audio.cpp**************************************************************************************
//*                                                  audioI2S-- I2S audiodecoder for ESP32,                                                              *
//********************************************************************************************************************************************************

#include "Arduino.h"
#include "WiFiMulti.h"
#include "src/Audio.h"

#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio* audio;
WiFiMulti wifiMulti;
String ssid =     "****";
String password = "****";

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    while(WiFi.status() != WL_CONNECTED){
        log_i("...");
        vTaskDelay(100);
    }
    audio = new Audio;
    audio->setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio->setVolume(12); // 0...21

audio->connecttohost("http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_6music/bbc_6music.isml/bbc_6music-audio=96000.norewind.m3u8");
// audio.connecttohost("https://jazzradio.ice.infomaniak.ch/jazzradio-high.aac");
// audio.connecttospeech("Hallo, wie geht es dir?", "de");
}

void loop(){
    if(audio) audio->loop();
    if(Serial.available() && audio){ // put streamURL in serial monitor
        String r=Serial.readString(); r.trim();
        log_i("r=%s", r.c_str());
        if(r == "i") log_i("inbuff filled =%i", audio->inBufferFilled());
        if(r == "a") audio->connecttohost("https://jazzradio.ice.infomaniak.ch/jazzradio-high.aac");
        if(r == "b") log_i("hallo");
        if(r == "d"){audio->stopSong(); delete audio; log_i("audio object destroyed"); }
    }
    vTaskDelay(1);
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
````

FatherMarco1971, 2024-10-29:
Just to confirm that my code work again after your commit.

Thanks a lot!

closed 2024-10-29 by FatherMarco1971