schreibfaul1/ESP32-audioI2S #877: Feature Request - independent L/R volume control

shinoryo1216, 2024-10-26:
First of all thank you again for the forceMono reactivation and all your work.

I tried to add this feature with the help of chatgpt and claude, but finally we gave up.
Is there a way to have a separate volume control? It would be useful for many cases such as making DIY speaker or radio box.

My plan: wire a tweeter to L channel and a woofer to R channel of a 2ch amplifier with one physical volume control (pam8403) playing mono radio streaming.

schreibfaul1, 2024-10-26:
You would have to implement this yourself according to your wishes. There is the function Gain, the variables (double)m_limit_left and m_limit_right control the volume, values between 0 (mute) and 1 (100%) are possible.

[image]

shinoryo1216, 2024-10-27:
Thank you very much for the advice.
Now I have a separate volume control ... only with stereo mode. Problem is, with my current code, it is placed before the forceMono mixdown, I've been working all day long and almost give it up.
I will get back to you when I find a solution.

shinoryo1216, 2024-10-27:
Now I reached a solution. My collaborators aka Chatgpt and Claude showed me very complicated ways, and now I found out that is ridiculously simple....

closed 2024-10-27 by shinoryo1216