schreibfaul1/ESP32-audioI2S #878: cs4344使用怎么配置

1359450511, 2024-10-28:
请问一下如果使用cs4344需要配置什么，cs4344可以使用吗，我现在使用默认配置只能听到一堆杂音

1359450511, 2024-10-28:
谁能告诉我

schreibfaul1, 2024-10-28:
[image]
The difference to the PCM5102A is that MCLK is used instead of BLCK.

1359450511, 2024-10-29:
我现在的接法和你的差不多，SCK这个没有用，使用的内部sck,请问软件里面需要特殊的配置么，还是说只需要配置引脚，我现在只能听到一堆杂音。我使用 5102 没有问题

schreibfaul1, 2024-10-29:
The only difference is the use of the MCLK to which you have to assign a pin. audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK); Unfortunately, BCLK must not remain free even if this pin is not used.
I use this board 
[image]

and have no problems with it. The output signal is slightly weaker than the 5102.

1359450511, 2024-10-30:
esp32_audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    esp32_audio.i2s_mclk_pin_select(I2S_MCLK);
    esp32_audio.setVolume(15);
    esp32_audio.connecttohost("https://ytcastmp3.radio.cn/60/stream_10414.mp3?type=1&key=70f3219358f2067927e55571b6f3a30d&time=67221d17");
这是我的初始化代码，我弄到现在依然没有任何声音，我看到控制台有打印出接收到的数据，控制台打印下面信息
[225964][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 209ms  0, 0, 0, 0, 16384, 15762, 15761, 11665, 7569, 3473
[226221][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[226477][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 232ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[226734][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[227008][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[227265][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[227521][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 232ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[227778][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[228052][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[228309][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[228554][D][Audio.cpp:3070] processWebStream(): ␛[31m2566ms no data received
[228566][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 16384, 16145, 16144
[228774][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 196ms  12048, 7952, 3856, 144, 143, 0, 0, 0, 0, 0
[229031][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[229305][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[229598][D][Audio.cpp:3070] processWebStream(): ␛[31m920ms no data received
[229599][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 270ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 16384
[229795][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 179ms  12288, 8192, 4096, 527, 526, 0, 0, 0, 0, 0
[230051][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 232ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[230325][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0
[230582][D][Audio.cpp:3075] processWebStream(): bytes available, 10 polls in 233ms  0, 0, 0, 0, 0, 0, 0, 0, 0, 0

1359450511, 2024-10-30:
#define I2S_DOUT 4
#define I2S_MCLK 3
#define I2S_LRC 6
#define I2S_BCLK 7
#define I2S_SCLK 5
Audio esp32_audio;
这是我的引脚配置，我现在还是没有任何声音 ， 我已经确认了接线，没有问题，功放也没有问题

1359450511, 2024-10-30:
已经找到原因了，使用的库不是最新版本，一直没发现这个问题，原来platformIO里面的库不是最新的版本，使用最新版本就解决了问题，能否把platformIO里面的库也同步为最新的

closed 2024-10-30 by 1359450511

schreibfaul1, 2024-10-30:
I don't know how PlatformIO synchronises the libraries.
You can enter the link in platfornio.ini and always have the "master"

````
lib_deps =
    https://github.com/schreibfaul1/ESP32-audioI2S.git
````