schreibfaul1/ESP32-audioI2S #879: 播放某些.mp3的网络流有些异常

caiyy, 2024-10-30:
播放某些.mp3的网络流有些异常
另外、从某个版本开始、ID3的中文信息解码也有问题、显示乱码、
希望有好心人可以解决一下、谢谢、

测试外网地址：
http://pal.caiyy.cn:28555/Music/EB高梦博-小灵通-小灵通-2023.mp3
http://pal.caiyy.cn:28555/Music/EB高梦博-死灰-杀死差不多先生-2015.mp3
http://pal.caiyy.cn:28555/Music/EB高梦博-中国嘻哈榜-Life Is Still a Struggle-2013.mp3
http://pal.caiyy.cn:28555/Music/EB高梦博-飘-飘-2022.mp3

http://192.168.1.218/mp3.php
info        buffers freed, free Heap: 164048 bytes
info        connect to: "192.168.1.218" on port 80 path "/mp3.php"
info        Connection has been established in 7 ms, free Heap: 163440 bytes
info        redirect to new extension at existing host "http://192.168.1.218/Music/EB高梦博-小灵通-小灵通-2023.mp3"
info        connect to: "http://192.168.1.218/Music/EB高梦博-小灵通-小灵通-2023.mp3"
info        The host has disconnected, reconnecting
mqtt received: CaiyyHome/HomeAudio/Mode - OFF
info        MP3Decoder has been initialized, free Heap: 132572 bytes , free stack 5180 DWORDs
lasthost    http://192.168.1.218/mp3.php
info        ID3 framesSize: 83846
info        ID3 version: 2.4
info        ID3 normal frames
id3data     Title: å°çµé
id3data     Artist: EBé«æ¢¦å
id3data     Album: å°çµé
id3data     Track: 1
info        Audio-Length: 6197416
info        Webfile: stream ready, buffer filled in 1712 ms
info        syncword found at pos 734
info        syncword found at pos 1537
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID FRAMEHEADER
info        syncword found at pos 878
info        syncword found at pos 0
info        MP3 decode error -9 : INVALID HUFFCODES
info        syncword found at pos 283
info        syncword found at pos 0
info        MP3 decode error -8 : INVALID SCALEFACT
info        syncword found at pos 90
info        syncword found at pos 0
info        MP3 decode error -9 : INVALID_HUFFCODES
info        syncword found at pos 455
info        syncword found at pos 0
info        MP3 decode error -6 : INVALID_FRAMEHEADER
info        syncword found at pos 133
info        syncword found at pos 0
info        MP3 decode error -9 : INVALID_HUFFCODES
info        syncword found at pos 1374
info        syncword found at pos 0
info        MP3 decode error -9 : INVALID_HUFFCODES
info        syncword found at pos 451
info        syncword found at pos 1161
info        syncword found at pos 226
info        syncword found

schreibfaul1, 2024-10-30:
Hello caiyy, 
I hope Chinese characters work now.
http://pal.caiyy.cn:28555/Music/EB高梦博-飘-飘-2022.mp3 returns:
[image]

caiyy, 2024-10-30:
@schreibfaul1 thinks

closed 2024-10-30 by caiyy