schreibfaul1/ESP32-audioI2S #883: realtime api playing really fast on esp32s3

devpras22, 2024-10-31:
is there a way to manually set the sample rate. the audio_play.loop(); is playing the audio at chipmunk speed.

schreibfaul1, 2024-10-31:
The sample rate specified in the header of the audio file is used. Please change it there.

devpras22, 2024-10-31:
even for raw pcm? it doesnt seem to be working with the headers :(

devpras22, 2024-11-01:
i have tried this res.setHeader('Content-Type', 'audio/wav; rate=24000; channels=1; bits=16'); and tried this -  res.setHeader('Content-Type', 'audio/wav');
   res.setHeader('Transfer-Encoding', 'chunked');
   res.setHeader('X-Audio-Sample-Rate', '24000');
   res.setHeader('X-Audio-Channels', '1');
   res.setHeader('X-Audio-Bits', '16');

devpras22, 2024-11-01:
so i think you meant wav headers not http headers..so i tried this as well.. but still playing audio super fast :( - if (delta && delta.audio) {
    const audioData = Buffer.from(delta.audio.buffer);
    if (audioStream) {
        const wavHeader = generateWavHeader(audioData.length, 24000, 1, 16);
        const wavChunk = Buffer.concat([wavHeader, audioData]);
        audioStream.write(wavChunk);
    }
}

schreibfaul1, 2024-11-01:
If you want to change the speed of an audio file, I recommend  Audacity. It's easier than editing an audio header by hand.

devpras22, 2024-11-01:
ugh? no no..I am able to save the file properly there is no need for audacity here...

I am having issues streaming the chunks to the esp32 in real-time.

Using whisper api, i had no issues with streaming chunks and playback on esp32s3 as the format was mp3. but the openai realtime api is currently in beta using onlyraw 16 bit PCM audio at 24kHz, 1 channel, little-endian. 

 The audio file saves correctly with this code -
 // Only save and finalize once per conversation to avoid multiple files
  if (item && item.status === "completed" && audioChunks.length > 0) {
    console.log("Conversation completed. Finalizing audio file.");

    const fullAudioData = Buffer.concat(audioChunks);
    const wavHeader = generateWavHeader(fullAudioData.length, 24000, 1, 16); // Ensure header matches PCM16 format
    const fullWav = Buffer.concat([wavHeader, fullAudioData]);

    // Check for existing files in the save path to avoid duplicate save
    const filePath = path.join(audioSavePath, `response_${Date.now()}.wav`);
    if (!fs.existsSync(filePath)) {
      fs.writeFileSync(filePath, fullWav);
      console.log(`Audio file saved successfully at ${filePath}`);
    } else {
      console.log("File already exists, skipping save to avoid duplication.");
    }

closed 2024-12-17 by github-actions[bot]