schreibfaul1/ESP32-audioI2S #884: Inconsistent streaming using connecttohost

MJenkings, 2024-11-01:
I have been using this library in a sketch for over a year. The hardware is an ESP32-S3 with sdcard and wifi connection. I play both MP3 and Internet streams. It has worked faultlessly to this point. Recently I upgraded my project to use the V3.0.7 ESP32 library.  I upgraded this library at the same time. 

While all the functions work, about 50% of the time the internet stream will fail to play sound. In the serial monitor it reports having started the stream and also pulls the stream data but no audio comes through. (I have tried several stream URLs, have very strong Wifi and a 1Gbps fibre internet connection)

On second or third attempt I get audio. I get no error messages when it fails because it seems to think it was successful. 

My audio chip is the MAX98357.  I'm no sure how to debug this or implement a fix as I cannot pin down the root cause.

schreibfaul1, 2024-11-01:
Decoding and feeding the I2S is handled by a separate task. This is omitted in audio.loop(), which runs too fast. To give background tasks more time, insert a vTaskDelay.

`````c++
void loop(){
    audio.loop();
    vTaskDelay(1);
}
````

MJenkings, 2024-11-01:
Thanks for the quick reply, unfortunately that hasn't solved it. I am noticing it does it occasionally while playing MP3s as well. It reports it opened the file, then immediately closed it. I don't want to increase the delay much further as I have a lot of other tasks in the main loop that run very quickly and would be slowed down by this. 

I'm wondering if it would benefit from being on it's own core but I am not sure how to move the audio to the 2nd core and still call it from code running on the first core. 

Mike Jenkings 

    On Friday 1 November 2024 at 18:06:20 GMT, Wolle ***@***.***> wrote:  
 
 

Decoding and feeding the I2S is handled by a separate task. This is omitted in audio.loop(), which runs too fast. To give background tasks more time, insert a vTaskDelay.
void loop(){
    audio.loop();
    vTaskDelay(1);
}
````
—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you authored the thread.Message ID: ***@***.***>

schreibfaul1, 2024-11-01:
I don't know the load distribution in your project. Originally the decoding in audio.loop() and therefore in the Arduino task was on core 1 by default. The internal audio task runs on core 0, you can change this with `audio.setAudioTaskCore(uint8_t coreID);`

closed 2024-12-16 by github-actions[bot]