schreibfaul1/ESP32-audioI2S #886: Opus player not working on some streams

marvinvis, 2024-11-02:
I have a working player. but some streams for Opus do not work:  This is using version 3.0.12

**this one is OK:** Knixx   |   http://s1.knixx.fm:5347/dein_webradio_512.opus
**this one is NOT** OK: Intense Radio - We love Dance   |   https://secure.live-streams.nl/opus.opus

See logging below:
Start playing Intense Radio - We love Dance. Memory: 235392
audio_info: buffers freed, free Heap: 238272 bytes
audio_info: connect to: "secure.live-streams.nl" on port 443 path "/opus.opus"
[125609][E][WiFiClient.cpp:320] setSocketOption(): fail on 0, errno: 9, "Bad file number"audio_info: SSL has been established in 439 ms, free Heap: 193992 bytes
audio_info: OPUSDecoder has been initialized, free Heap: 193160 bytes , free stack 3372 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
[126561][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126575][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126588][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126602][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126616][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126630][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126644][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126658][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126672][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126686][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126700][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126714][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126728][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126742][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126756][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126770][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126783][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126797][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126811][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126825][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126839][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126852][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126865][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126878][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126892][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126906][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126920][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126934][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126948][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126962][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126976][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[126990][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127003][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127017][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127030][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127044][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127058][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127072][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127086][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127099][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127112][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127125][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127138][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127152][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127165][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127178][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127192][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127205][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127218][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127231][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127244][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127257][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127270][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127284][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127298][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127311][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127324][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127338][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127352][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127366][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127379][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127393][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127406][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127420][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127434][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127447][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127460][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127474][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127488][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127501][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127514][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127527][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
button pressed
[127541][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127555][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
Start playing Knixx. Memory: 193160
[127568][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127581][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127594][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
[127607][E][opus_decoder.cpp:206] opusDecodePage0(): OpusHead not found
audio_info: buffers freed, free Heap: 237668 bytes
audio_info: connect to: "s1.knixx.fm" on port 5347 path "/dein_webradio_512.opus"        
audio_info: Connection has been established in 21 ms, free Heap: 237668 bytes
audio_info: OPUSDecoder has been initialized, free Heap: 235280 bytes , free stack 3372 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 48000
audio_info: BitsPerSample: 16
audio_info: BitRate: 170133

schreibfaul1, 2024-11-02:
Yes, that's a mistake, I'll have a look at that in the next few days
[image]

marvinvis, 2024-12-14:
Hi Wolle, I would like to know if there is anything I could do to help in fixing this issue.

kind regards,
Marvin

schreibfaul1, 2024-12-14:
Hello marvinvis,
I have completely lost sight of this topic. In the OPUS head, 44100 Hz is specified as the sample rate, which is unusual. I have corrected this. However, the system switches from CELT to SILK during pauses. The SILK decoder works, but the "transition", i.e. the switchover from CELT to SILK, is not yet complete. This will not be ready until next year.

closed 2025-01-30 by github-actions[bot]