schreibfaul1/ESP32-audioI2S #887: Cracking noises and some playbacks stop

foorschtbar, 2024-11-02:
After switching from Release 3.0.8 to 3.0.12 i had two problems:
- Playback for some mp3 stops after a few seconds
- I hear a random cracking noise - 3.0.8 had also some cracking noise, but very very quiet, with 3.0.12 it is very loud.

Nothing special in the debug output. Anyone has the same problem?

PS: @schreibfaul1 awesome library!

dominickairmyne, 2024-11-05:
I have been having alot of issues with this as well!

ra0943-VK3ACH, 2024-11-12:
Im also having issues with this. Would you be able to share your code? and are you playing local files or streaming them

schreibfaul1, 2024-11-12:
In the old version, the audio.loop() did everything. Since the WiFiClient is not blocking-free, I have moved the part responsible for decoding and I2S "feeding" to an extra task. This way many streams, especially m3u8, run more smoothly. 
This relieves the audio.loop(), which runs on core 1 by default. It may be that there is not enough time for other tasks, in which case a vTaskDelay() 1...10ms provides a remedy.

[image]

Andernfalls kann es sein, dass in manchen Projekten jetzt Kern 0 zu stark belastet wird, z:B durch eigene Tasks, dann kann der Audiotask von Kern1 nach Kern 0 verlagert werden.

ra0943-VK3ACH, 2024-11-13:
Adding vTaskDelay has solved the issue for me! Thankyou!

foorschtbar, 2024-11-13:
Yes, the `vTaskDelay(1);` fixed the crackings also for me, thank u.

But i still need to downgrade to 3.0.8 to play a mp3 that stops playing (or is quiet until the end)  with 3.0.12. No useful information with `audio_info`. And the end of the file, you could here something and then `audio_eof_mp3` is called.

closed 2024-12-29 by github-actions[bot]