schreibfaul1/ESP32-audioI2S #888: more a question than an issue

philippedc, 2024-11-03:
Hi, I'm finally using Audio.h v2.0.6 as Expressif Sytems (ESP32) version 2.0.3  is installed. I've tried Audio.h v3.0.0 but with this version I could not operate properly with audio_eof_speech nor audio_eof_mp3.

I have this matter for instance with this station: https://stream.zeno.fm/f38bxpt3v2quv
where I get HTTP/1.1 401 message in showstreamtitle

I think it is a problem with redirect url.
My question is: how can I get on console all AUDIO_INFO messages present in Audio.cpp ?

Thanks

schreibfaul1, 2024-11-03:
Hi philippedc,

thrURL https://stream.zeno.fm/f38bxpt3v2quv redirects to https://stream-155.zeno.fm/f38bxpt3v2quv?zt=eyJhbGciOiJIUzI1NiJ9.eyJzdHJlYW0iOiJmMzhieHB0M3YycXV2IiwiaG9zdCI6InN0cmVhbS0xNTUuemVuby5mbSIsInRtIjpmYWxzZSwicnR0bCI6NSwianRpIjoiRjZiVFppcGxSbFNZMUNTUDkwMFNxdyIsImlhdCI6MTczMDY2OTcxOSwiZXhwIjoxNzMwNjY5Nzc5fQ.yL97Tlv8xqsRbGD1e575rS5Yph2MEMQUgm-WnP4qjKI"
that is 296 characters, previously it was separated by 256 characters. This is no longer the case with the new version.

But nothing has changed in the eof events. I don't know exactly where the problem lies.

In the example, all audio files on the SD card in the /mp3 folder are played one after the other.
````c++
#include "Arduino.h"
#include "Audio.h"
#include "SD_MMC.h"
#include "FS.h"

#define I2S_LRC     26
#define I2S_DOUT    25
#define I2S_BCLK    27

#define SD_MMC_D0   2
#define SD_MMC_CLK  14
#define SD_MMC_CMD  15

void getNextFile(); //proto

Audio audio;

File dir;
const char audioDir[] = "/mp3";

void setup() {
    Serial.begin(115200);
    pinMode(SD_MMC_D0, INPUT_PULLUP);
    SD_MMC.setPins(SD_MMC_CLK,SD_MMC_CMD, SD_MMC_D0);
    if(!SD_MMC.begin( "/sdmmc", true, false, 20000)){
        Serial.println("Card Mount Failed");
        return;
    }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, 0);
    audio.setVolume(17); // 0...21 Will need to add a volume setting in the app
    dir = SD_MMC.open(audioDir);
    getNextFile();
}

void loop(){
    audio.loop();
    vTaskDelay(1);
}

void getNextFile(){
    File file = dir.openNextFile();
    if(!file) return;
    Serial.printf("next file: %s\n", file.path());
    String audioFile = file.path();
    file.close();
    audio.connecttoFS(SD_MMC, audioFile.c_str());
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
    getNextFile();
}

````

philippedc, 2024-11-05:
Hi Wolle, many thanks for your answer, I'll be doing test again. And test SDMMC too.

how can I get on console all AUDIO_INFO messages present in Audio.cpp. There are not displayed by default.

schreibfaul1, 2024-11-06:
The output depends on the TAGs in the header of the audio file and how the decoder is set. A typical example looks like this: 
[image]

philippedc, 2024-11-08:
Excuse me, I was speaking about AUDIO_INFO that should display console message, for instance:

`
      if(specialIndexOf(data, "ID3", 4) != 0) { // ID3 not found
            if(!m_f_m3u8data) AUDIO_INFO("file has no mp3 tag, skip metadata");
            m_audioDataSize = m_contentlength;
            if(!m_f_m3u8data) AUDIO_INFO("Audio-Length: %u", m_audioDataSize);
            return -1; // error, no ID3 signature found
        }
`

closed 2024-12-24 by github-actions[bot]