schreibfaul1/ESP32-audioI2S #891: crash with new version

Arielhh, 2024-11-04:
when using the official version 3.0.12 it works fine but with new versions it crashes.  
This station requires a large buffering. with 3.0.12 the buffering memory is not forced to be empty. with new versions the memory is forced to be empty and crashes that station. 
They are all links to the same station. It happens also with other stations connected to the same servers.

https://playerservices.streamtheworld.com/api/livestream-redirect/ARN_WSFM.mp3
https://ais-arn.streamguys1.com/au_002/playlist.m3u8
https://ais-arn.streamguys1.com/au_002_icy
https://playerservices.streamtheworld.com/api/livestream-redirect/ARN_WSFMAAC.aac

schreibfaul1, 2024-11-04:
The wifiClient is not non-blocking. I have therefore provided a separate task for the decoders. This stack (3300 * 4 bytes) reduces your available memory. To compensate, you can reduce the Arduino stack.

Please write before the setup()   `SET_LOOP_TASK_STACK_SIZE(8192 -3300)`
8129 is the predefined value and you can reduce the stack as long as your programme is running well.

Arielhh, 2024-11-05:
Thanks, I will try it and update.

Arielhh, 2024-11-05:
I tried various values, it is still behaving the same.

closed 2024-12-21 by github-actions[bot]