schreibfaul1/ESP32-audioI2S #893: "standalone" tone generator does not work

nightflyer88, 2024-11-04:
Hello

Cool library with lots of functions! The examples worked straight away. Very good work!

Now I want to make a simple sine wave sound generator, without a music stream from another source. I tried this example:
```
void audio_process_i2s(int16_t* outBuff, uint16_t validSamples, uint8_t bitsPerSample, uint8_t channels, bool *continueI2S){
    int16_t sineWaveTable[44] = {
         0,   3743,   7377,  10793,  14082,  17136,  19848,  22113,  23825,  24908,
      25311,  24908,  23825,  22113,  19848,  17136,  14082,  10793,   7377,   3743,
         0,  -3743,  -7377, -10793, -14082, -17136, -19848, -22113, -23825, -24908,
     -25311, -24908, -23825, -22113, -19848, -17136, -14082, -10793,  -7377,  -3743
    };

    static uint8_t tabPtr = 0;
    int16_t* sample[2]; // assume 2 channels, 16bit
    for(int i= 0; i < validSamples; i++){
        *(sample + 0) = outBuff + i * 2;     // channel left
        *(sample + 1) = outBuff + i * 2 + 1; // channel right

        *(*sample + 0) = (sineWaveTable[tabPtr] /50 + *(*sample + 0));
        *(*sample + 1) = (sineWaveTable[tabPtr] /50 + *(*sample + 1));
        tabPtr++;
        if(tabPtr == 44) tabPtr = 0;
    }
   *continueI2S = true;
}
```

The sine wave sound only works if another stream (for example: `audio.connecttoFS(SD, "ACDC - Back In Black.mp3"`) is already running, and is mixed into the music accordingly. Without a stream from the SD card, no sine wave sound is output.

Is there a way to start an "empty" stream?

schreibfaul1, 2024-11-04:
Hello,
that's not what the library is for. The example is only intended to show how to add an external audio source.
You could create a sound with Audacity and export it. However, the file is finite, but you could make it very long.

nightflyer88, 2024-11-04:
Hallo Schreibfaul !

Ok ich verstehe. Gibt es eine bessere Möglichkeit ? Wie muss ich die Lib anpassen ?

Ich habe versucht mit:

```
audio.setSampleRate(44100);
audio.setBitsPerSample(16);
audio.setChannels(2);
audio.setBitrate(128000);
//audio.playAudioData();
audio.playChunk();
```
Die Funktionen entsprechend auf Public gestellt, funktioniert jedoch nicht... es braucht anscheinend noch etwas mehr dazu

closed 2024-11-07 by nightflyer88