schreibfaul1/ESP32-audioI2S #895: Problem ! User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/129.0.0.0 Safari/537.36\r\n");

bhpjean, 2024-11-05:
This entry:

strcat(rqh, "User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/129.0.0.0 Safari/537.36\r\n");

This causes servers based on SHOUTcast server software to reject connections from radios based on the new library.
I think the name is probably too long. I don't know.

I changed this entry in Audio.cpp to the old one:

strcat(rqh, "User-Agent: ESP32 audioI2S\r\n");

and the radio tuner connects normally to the server based on SHOUTcast server software

check these streams and you will see:

http://relay1.slayradio.org:8300/
http://stream3.polskieradio.pl:8950

Also, great respect for the work in writing the esp32 i2s audio library

Best regards

Piotr D

Marcin-Fenger, 2024-11-05:
User-Agent: iTunes/12.6.5 (Macintosh; OS X 11.0) AppleWebKit/611.2.7.1.4 - works
User-Agent: VLC/3.0.21 LibVLC/3.0.21  - works

I think it should be changed.

schreibfaul1, 2024-11-06:
Thanks for the good work. I have changed the user agent.

closed 2024-12-22 by github-actions[bot]