schreibfaul1/ESP32-audioI2S #896: Still crashes on esp32s3，esp32duino ver 3.0.7

ChinaBurn, 2024-11-06:
with Version 3.0.13j，Has the constructor for initializing the library been corrected? Previously, I saw that the problem was caused by the initialization of the library before the psram.

schreibfaul1, 2024-11-06:
Yes, the PSRAM is now initialised later, together with `setPinout()`.
A developer from Espressif also said that initialising the PSRAM in the constructor is not a good idea.

ChinaBurn, 2024-11-07:
Did you initialize the 3.0.3j version in esp32duino normally? The reason for my crash is the same as before, PSRAM problem

tv4you2016, 2024-11-07:
https://github.com/schreibfaul1/ESP32-audioI2S/commit/98488c5e1a8d10fc257c12cd80bc52084e253b06#diff-6033949d768051c96b2a380edec84a3e1cbe58b3a94e41aae408c6a8f8bbc2b2R11

try to rollback this commit

schreibfaul1, 2024-11-08:
Hello tr4you2016,
since Arduino V3.0.7 it is no longer possible to initialise memory areas in the PSRAM without restrictions.

````c++
char* buff = NULL;
bool psramFound = psramInit();
if(psramFound) buff = (char*)  ps_malloc(2048);
else  buff = (char*) malloc(2048);
if(!buff) log_e("oom");
````
then is psramFound == true und buff is NULL
No idea why, maybe SPI is not ready yet.

Now I initialise the buffers in the constructor with malloc and later I release the RAM and initialise with ps-malloc.

ChinaBurn, 2024-11-11:
It still causes esp32s3 to restart infinitely, and looking at the debug information, everything was normal before until setPinout was called, and no debug information was output.  Audio Ver is 3.0.13q

schreibfaul1, 2024-11-11:
If there is no debug information at all, one of the assigned pins is probably invalid, 

Example for S3
#define I2S_DOUT   9
#define I2S_BCLK    3
#define I2S_LRC      1

ChinaBurn, 2024-11-12:
> If there is no debug information at all, one of the assigned pins is probably invalid,
> 
> Example for S3 #define I2S_DOUT 9 #define I2S_BCLK 3 #define I2S_LRC 1

You are right, I changed the pins and it worked. The strange thing is that my previous pins were:
#define I2S_DOUT 37
#define I2S_BCLK 36
#define I2S_LRC 35
This was completely normal before.

closed 2024-11-12 by ChinaBurn