schreibfaul1/ESP32-audioI2S #902: Problem with a certain FLAC stream

bhpjean, 2024-11-10:
Dear author of the i2s_audio library, I have a problem and I can't explain it. Maybe your knowledge of this problem will explain to me where the problem is.

I turn on the stream: http://stream.radioparadise.com/flac
I listen for a while, then I switch to another FLAC stream, e.g. this one: http://stream.radioparadise.com/eclectic-flac and I also listen to it for a while and it goes back to http://stream.radioparadise.com/flac
and I do not receive information about the codec, after which ESP restarts.

Basically, this happens every time I switch from any FLAC stream to the Radioparadise stream. I understand that only Radioparadise is a problem here, I have not found any other flac station with such problems. My knowledge in this area is too small to cope with finding the problem. Perhaps it is about some time interval to extend the waiting time for the library to recognize the codec.

Best regards, Piotr

Arielhh, 2024-11-10:
I agree flac is not playing right.
In the meantime you can use other formats : https://radioparadise.com/listen/stream-links

bhpjean, 2024-11-11:
It plays correctly the first time it is called, the next time it causes the ESP to restart

bhpjean, 2024-11-11:
> I agree flac is not playing right. In the meantime you can use other formats : https://radioparadise.com/listen/stream-links

It plays correctly the first time it is called, the next time it causes the ESP to restart

schreibfaul1, 2024-11-11:
It looks like a timing problem, the data is available before the FLAC decoder is fully initialised.
The reason for this is that with ogg the first block has to be analysed first to determine whether it is OPUS, FLAC or VORBIS. I think it works better now.

Arielhh, 2024-11-11:
Thanks for the fix, it is much better now, the response and uploading of the station in much faster,  but the sound is still a little bit choppy :(

closed 2024-12-27 by github-actions[bot]