schreibfaul1/ESP32-audioI2S #903: 小米手机自带的音乐文件怎么发送到DLNA

1359450511, 2024-11-11:
就是如何进行DLNA播放，就像手机上的视频可以推送到电视机播放一样，手机的音乐怎么通过DLNA推送到ESP32-I2S audio

schreibfaul1, 2024-11-11:
Exactly, if the URL is known, a song can be played from a DLNA server.
You can find out the URL with the VLC Player.
[image]
Or you can use a library to display the contents of the DLNA https://github.com/yellobyte/SoapESP32

foorschtbar, 2024-11-13:
> 就是如何进行DLNA播放，就像手机上的视频可以推送到电视机播放一样，手机的音乐怎么通过DLNA推送到ESP32-I2S audio

if the issue were in english, then others with the same question would also benefit...

schreibfaul1, 2024-11-14:
This library requires a valid URL. The URL can be found out with the VLC Player, as I have described. To use the full DLNA functionality (title, album genre...) another library like the one from Yellbyte is required.

closed 2024-12-31 by github-actions[bot]