schreibfaul1/ESP32-audioI2S #908: playChunk()  - 16bit 1 ch sound distorted

Marcin-Fenger, 2024-11-14:
There is something wrong whith 16bit 1ch streams - https://rs6-krk2.rmfstream.pl/RMFFM48

In previous version (3.0.13j) validSamples was 2024, but now after changes(16bit conversion) is 1024 and sound is distorted.

2ch plays without problems with 1024 or 1152 validSamples.

Plaese take a look and check if this channel plays ok by your side.

Marcin-Fenger, 2024-11-14:
I have tested sinus 1kHz mp3 on oscilloscope and it is distorted. On old release works fine. I have already spent 2 days to solve that without succes.

schreibfaul1, 2024-11-14:
Hello Marcin-Fengler,
thank you for your contribution. I recently changed the routine for mono. I think it works better now.
However, https://rs6-krk2.rmfstream.pl/RMFFM48 does not really transmit in mono but in parametric stereo. The RAM in the ESP32 is limited and decoding must take place in the PSRAM. Unfortunately, the "quad" PSRAM is too slow for this, so Parametric Stereo is switched off. PS is possible with "octal" PSRAM in the ESP32-S3 N16R8 or N8R8.

bhpjean, 2024-11-14:
Hello schreibfaul1
I'll join the topic. After the latest changes in the latest library, some mono stations are almost unlistenable, I was curious what happened. I played the sine test and got scared. This is what my oscilloscope shows. The sine wave is terribly jagged..

[image]

In the i2s_audio 3.0.13j library version, the sine looks very good.

However, it might be worth returning to the previous conversion method.

[image]

Hello...
I'll join the topic. After the recent changes in the latest library, some mono stations are almost unlistenable, I was curious what happened. I played the sine test and got scared. This is what my oscilloscope shows. The sine wave is terribly jagged..

In the i2s_audio 3.0.13j library version, the sine looks very good.

However, it might be worth returning to the previous conversion method.

Thank you for improving the decoding of FLAC streams. In the current library, it works very well.

Hello...
I will join the topic. After the recent changes in the latest library, some mono stations are simply unlistenable, I was curious what happened. I played the sine test and got scared. This is what my oscilloscope shows. The sine wave is terribly jagged...

In the i2s_audio 3.0.13j library version, the sine looks very good.

However, it might be worth returning to the previous conversion method.

Thank you for improving the decoding of FLAC streams. In the current library, it works very well.

Best regards

Piotr

Marcin-Fenger, 2024-11-15:
1kHz

[image]

10kHz - more distorted

[image]

Signals looks like amplitude modulated with second frequency

schreibfaul1, 2024-11-16:
Hello, 
I2S is supposed to be able to play mono on two channels. It looks like this only works with the ESP32-S3. I have restored the old state and double the channels as before.

Marcin-Fenger, 2024-11-16:
Thank you very much. Now all sound perfect.

I play with you library for years. I very appreciate your work. 

I have some improvements to consider.

1. Place loop in RTOS task pinned to core1 with prioryty 1, stack 10000 for priority controll
2. Disable loop
void loop()
{
	vTaskDelete(null);
}
3.  Comment line 5729 in audio.cpp to disable  printing chunk size on serial m_f_log when stream is playing
4.  Change ps_mallosc to malloc for m_outBuff. Vu & 3 x fir and I2S have to work on PSRAM. It takes time with high usage SPI. I think it will be better when that buffer stays at RAM. I have already chenged it. 
5. In the I2S config I set APLL_ENABLED. Mclk is more stable. I use PCM1796. It works fine, but should set  m_i2s_config.fixed_mclk  = false;  m_i2s_config.mclk_multiple = I2S_MCLK_MULTIPLE_256. The same config is for PCM 1794 and PCM5102 if You would like to use mclk. Piotr uses this clock line with PCM5102 in his setup.

BTW, I use loop on core 1, prioryty 1 and audio task on core 1 prioryty 2 as you set :)

Greetings!

Marcin-Fenger, 2024-11-16:
And new observation the 3.0.13t

ADTS 48k cause very high duty of processor. Scroll stops on core 0 on this stream https://stream320.radiodzwiek.pl.
Was normal, but now duty is very very high. Did You change something in this codec? In versions j & r was good.

schreibfaul1, 2024-11-17:
Hi Marcin,
thanks for your feedback. I will see what I can implement from your suggestions.
What would be problematic is to remove audio.loop() from the Arduino loop task. Then an intertask communication (message queues) would be necessary for the further processing of events like audio_streamtitle(), which is difficult for beginners.

https://stream320.radiodzwiek.pl is demanding with a bit rate of 384K (stream4.nadaje.com). However, this is possible if you increase the number of TCP connections and the RX buffers in menuconfig lwip/TCP.

Marcin-Fenger, 2024-11-17:
Hi

You are right. You have  xSemaphore in audioConnect, so implemet xQueue is only question of time :). You can place xQueuesend with hostname and xQueueReceive inside audioTask or directly inside library. It will be work in any task.

Thanks for the tip of buffer.

My config is now:
#define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 16
#define CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM 16
#define CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM 128
#define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM 64
#define CONFIG_ESP32_WIFI_RX_BA_WIN 32
#define CONFIG_ESP32_TCP_SND_BUF_DEFAULT 65
#define CONFIG_ESP32_TCP_WND_DEFAULT 65
#define CONFIG_ESP32_WIFI_DYNAMIC_RX_MGMT_BUFFER 1
#define CONFIG_ESP32_WIFI_IRAM_OPT 1
#define CONFIG_LWIP_IRAM_OPTIMIZATION 1
#define CONFIG_ESP32_WIFI_EXTRA_IRAM_OPT 1

What do You think?

Greetings!

closed 2025-01-01 by github-actions[bot]