schreibfaul1/ESP32-audioI2S #911: Buffers ara not complitly cleared

Marcin-Fenger, 2024-11-21:
Release 3.0.13t

When volume set to 0 and I do connect to next station than time to time radio plays previous station for the short time. It looks like uncleared buffer or something like that. It is very curious, because when vol is set to 0 it should not have been sent data to i2s in level higher than 0. Gain in playChunk() reduce the sample values. I am not sure if setDefaults() works perfect.

vol 0 was for test only. I have observed it while working on fadein and fadeout functions.

It is not regular. It occures time to time but it is very annoying.

In this case i tested:
https://listen.181fm.com/181-chilled_128k.mp3
http://relay.181.fm:8004/

schreibfaul1, 2024-11-22:
I think it was because the read pointer of the InBuff changed after stopSong().

Marcin-Fenger, 2024-11-22:
I have checked v. 3.0.13u out and issue still occures.

 vol = 6

Look at that

[image]

8.7ms peak in high vol play first and after normal sound of vol 6.

Marcin-Fenger, 2024-12-03:
Problem is not solved yet; I've tested on 3.1@idf5.3 and peak of previous sound is occured; I will manage with it using programabe(i2c) pcm1796 or by next DSP but I think this issue should be resolve. Library 3.0 without RTOS audio task is free of it.

closed 2025-01-18 by github-actions[bot]