schreibfaul1/ESP32-audioI2S #912: SSL - Memory allocation failed

Father001, 2024-11-21:
### How to free up memory？？？
I'm using the code for this example： [https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/separate_audiotask/separate_audiotask.ino](url)

and I add one button， when button clicked, I called `audio.stopSong(); `, and use httpClient post new url, but i get this error:

[288634][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
[288635][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512

What is the function of audio.stopSong()? it no free up memory?

schreibfaul1, 2024-11-22:
The WiFiClientSecure releases the memory itself. Alternatively, you can move mbedTLS to the PSRAM if there is a lack of memory.

Father001, 2024-11-22:
> The WiFiClientSecure releases the memory itself. Alternatively, you can move mbedTLS to the PSRAM if there is a lack of memory.

Thank you for the answer,！I solved the problem，

closed 2025-01-06 by github-actions[bot]