schreibfaul1/ESP32-audioI2S #916: Metadata encoding

Pako2, 2024-11-23:
First of all, I have to thank you for the great library, without which I can't imagine working on my project.
I'm using the stable version 3.0.12.
I've encountered one problem with ID 3 tags of MP3 files, if the metadata contains diacritics.
If I use the metadata editor to save the ID3 tags as ID3v2.3, everything is fine and the ESP32-AudioI2S library decodes the metadata correctly.
However, if I save the ID3 tags using the newer version ID3v2.4, the metadata is garbled.
Maybe I'm doing something wrong?

schreibfaul1, 2024-11-23:
Yes, there are differences in text encoding between ID3v2.3 and ID3v2.4. Can you please attach a short mp3 file here that has this problem.

Pako2, 2024-11-24:
To avoid copyright infringement, I used a test file I created some time ago. It contains only silence (five minutes long).
I write the metadata either with [Mp3tag](https://www.mp3tag.de) or [TagScanner](https://www.xdlab.ru/en/). But it doesn't matter, both give the same results.
[AudioSamples.zip](https://github.com/user-attachments/files/17892579/AudioSamples.zip)

Pako2, 2024-11-24:
Ah, thanks for the info.
So I have to use a newer version than the official 3.0.12.
I'm not very happy about that, because I'm having a lot of problems with it (within my rather complex project) ...
When trying to connect to a stream of type ".aac", the ESP32 resets with the error
`assert failed: block_locate_free heap_tlsf.c:441 (block_size(block) >= size)`
I've experimented with _SET_LOOP_TASK_STACK_SIZE(8192 - 3300);_ , but that doesn't help much ...

For audio I use a separate task:
`xTaskCreatePinnedToCore(audioTask, "audio", 6000, NULL, 2, &Task0, 0);`

Excerpt from my platformio.ini :
```
board = esp-wrover-kit
build_flags =
          -D BOARD_HAS_PSRAM
```
Do you have any ideas on how to proceed now?

schreibfaul1, 2024-11-24:
Memory space is scarce for larger projects. It helps if you move lwip/mbedTLS to PSRAM. To do this, you can change some settings in platformio.ini to enable menuconfig.
````
[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip  ;Arduino 3.0.5 / IDF 5.1.4
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc1/platform-espressif32.zip ;Arduino 3.1.0 / IDF 5.3.0
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip   ;Arduino core 3.0.7 / IDF 5.1.4

framework = arduino, espidf

build_flags     = .......

board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

lib_deps =
    https://github.com/schreibfaul1/ESP32-audioI2S.git
````

Pako2, 2024-11-24:
Thanks for the advice, I'll try it.
But I'm not sure if it will have any effect in my case.
lwip is definitely used, but I don't think I need mbedTLS in my project.
Will it help save some memory anyway?

schreibfaul1, 2024-11-24:
no https ?

If you have an extra audio task, and audio.loop() is alone in the task, please add a vTaskDelay(1) so that other tasks get computing time.

Pako2, 2024-11-24:
> please add a vTaskDelay(1) so that other tasks get computing time

Of course I have that. Without it, it certainly wouldn't work even with version 3.0.12.
The problems `assert failed: block_locate_free heap_tlsf.c:441 (block_size(block) >= size)` only appeared with version 3.0.13j (and higher), which I tried because of a fixed bug with ID3v2.4.

Marcin-Fenger, 2024-11-25:
> platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip  ;Arduino 3.0.5 / IDF 5.1.4

platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc3/platform-espressif32.zip ;Arduino 3.1.0 / IDF 5.3.1+

works - just play !!!

Pako2, 2024-11-29:
Thank you for your well-intentioned advice.
I've been working on my project for a long time and when I started, Arduino was in version 2.0.
So now I had to migrate from 2.0 to 3.0 - I guess it was time.
But the migration was not an easy job at all. I came across quite a few things that prevented compilation and had to be solved manually.
But the biggest problem was ESPAsyncWebServer. It had a lot of problems during compilation and so I used [the upgraded version from mathieucarbou](https://github.com/mathieucarbou/ESPAsyncWebServer).
The compilation went well, but every time immediately after opening the websocket there was a crash with the error `CORRUPT HEAP: Bad tail at ... Expected 0xbaad5678 got 0xbaad5600`.
I struggled with this for many hours, but in vain. Finally I came to the conclusion that the cause was in the new ESPAsyncWebServer library.
So I went back to the [original version from me-no-dev](https://github.com/me-no-dev/ESPAsyncWebServer) and manually fixed the issues that were preventing compilation.
And lo and behold - it works again, but now with Arduino 3.0!
So for now, the "Metadata encoding" issue can be considered closed. Thanks to all contributors.

closed 2024-11-29 by Pako2