schreibfaul1/ESP32-audioI2S #918: Internet radio station decoded fail(something like "AAC decode error -3 : INVALID_ADTS_HEADER")

mifarelight, 2024-11-27:
Thank you for this great project!

I try to play online m3u8 url stream stations. Some of them work well, but another failed.

for example, url: https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio.m3u8

```
info        Connect to new host: "https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio.m3u8"
info        buffers freed, free Heap: 92136 bytes
info        SSL has been established in 504 ms, free Heap: 48896 bytes
info        new request: "https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio-1730328655.ts?txspiseq=some...number"
info        AACDecoder has been initialized, free Heap: 48656 bytes , free stack 8204 DWORDs
info        syncword found at pos 45
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
...
info        syncword found at pos 26
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 48000
info        BitsPerSample: 16
info        BitRate: 74625
info        AAC HeaderFormat: ADTS
info        AAC Codec: MPEG-4 LowComplexity
info        AAC decode error -15 : NCHANS_TOO_HIGH
info        syncword found at pos 8
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 171
info        syncword found at pos 0
info        AAC decode error -15 : NCHANS_TOO_HIGH
info        syncword found at pos 3
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 3
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 165
```

Any suggestions would be greatly appreciated.

schreibfaul1, 2024-11-28:
I am not able to reproduce the error. However, `free heap: 48656 bytes` seems very small to me.

mifarelight, 2024-11-29:
I got in touch with ESP32-audioI2S from this project:
https://github.com/moononournation/LVGL_Music_Player.git (https://www.instructables.com/Design-Music-Player-UI-With-LVGL/)

This project uses ESP32-audioI2S to play audio files saved in the SD card and uses LVGL to implement a classic WinAMP user interface.

I simply modified it to connect and play online radio stations. Since I am not familiar with FreeRTOS/ESP32-audioI2S/LVGL..., decoding failures, online playback sound freezes, and user interface crashes and restarts often occur during the program running.

I hope to gradually become familiar with and master RTOS programming, music decoding and playback through such applications. I suspect that the memory resource allocation is not appropriate, and there may be memory leaks in the program. I try to find a way to troubleshoot the problem and improve it.

You mentioned that the "free heap" seems very small, and I am not sure where to modify it so that the ESP32-audioI2S task can allocate more memory space.

Thank you for your attention and reply.

mifarelight, 2024-11-29:
The model of ESP32 I am using is ESP32-S3-WROOM-1-N8R2 with 2MBytes external RAM.

During the initialization process of ESP32-audioI2S task, I selected the following parameters. I wonder if they are appropriate?
```c
audio.setBufsize(32*1024, 1024*1024);
```

mifarelight, 2024-11-30:
Since you said that the free heap is a bit low, I applied(by calling heap_caps_malloc MALLOC_CAP_SPIRAM) for the buffer used by the UI to the PSRAM, so there is much about 30kBytes(from 48896 to 79496) of free heap.

However, the decoding failure prompt("AAC decode error") did not change. When I played other online radio stations that could be decoded normally or even played mp3 files on the SD card, the whole program still restarted after running for a while. I can't find a way to troubleshoot the problem.

```
Play: https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio.m3u8
info        Connect to new host: "https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio.m3u8"
info        buffers freed, free Heap: 122140 bytes
info        SSL has been established in 451 ms, free Heap: 79496 bytes
info        new request: "https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio-1730409961.ts?txspiseq=some...number"
info        AACDecoder has been initialized, free Heap: 79252 bytes , free stack 5900 DWORDs
info        syncword found at pos 45
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        new request: "https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio-1730409962.ts?txspiseq=some...number"
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        AAC decode error -3 : INVALID_ADTS_HEADER
info        syncword found at pos 0
info        new request: "https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio-1730409963.ts?txspiseq=some...number"
info        AAC decode error -3 : INVALID_ADTS_HEADER

```

mifarelight, 2024-12-04:
I found the cause of the problem.

After several days of trying, the decoding error still exists.

So I changed my mind and tried to use ESP32-S3 to drive W5500 PHY, that is, to connect to the network by wire, and then the radio station of this URL can be played normally.

Therefore, the root of the problem is that the network quality of the Wi-Fi I use is not good.

Thank you again for your contribution of this great project.

closed 2024-12-04 by mifarelight